bigtreetech/BIGTREETECH-TouchScreenFirmware #2604: [Q] Does it work with ELEGOO Neptune 2S?

Johnnii360, 2022-09-19:
Does this firmware also work with the ELEGOO Neptune 2(S)?

My model has a MKS board clone the ZNP Robin Nano v1.3 and MKS display clone ZNP Robin TFT35 v1.0. In the description is MKS TFT35 v1.0 listed but I don't know if this is also the case fpr ZNP TFT35 v1.0? I don't want to kill my board again. :)

I hope it's also based on Marlin 2.1.x?

rondlh, 2022-09-22:
Can you send some detailed pictures of the front and back of the display?

Johnnii360, 2022-09-23:
Sure. These pictures are not made by me but it's the same.

[image]
[image]

rondlh, 2022-09-23:
Thanks, looks like an SPI TFT to me, which is fully controlled by Marlin, I found the printer firmware source here:
https://github.com/manniefesto/neptune-2-firmware
Which configures the display is a: MKS_ROBIN_TFT35

This means this display is NOT supported by the firmware here, but it's fully supported by Marlin.

closed 2022-12-04 by stale[bot]