bigtreetech/BIGTREETECH-TouchScreenFirmware #2609: [FR] Enhancement to #2575 (function hints)

rondlh, 2022-10-03:
#2575 is a useful enhancement that provides a history function and clear line function to the console.
But unfortunately most users will not be aware of the added functionally, a hint can solve this issue.

Here my suggestions:
```
      lastIndex = nowIndex;  // update gcode size
      if (gcodeBuf[0] == 0) 
		drawGcodeText("Prev  |  Next  |  Clear");
	  else
	    drawGcodeText(gcodeBuf);
    }

    loopBackEnd();
  }

  // restore default
  GUI_RestoreColorDefault();
}
```

These 3 lines were added:
```
      if (gcodeBuf[0] == 0) 
		drawGcodeText("Prev  |  Next  |  Clear");
	  else
```

This will show a hint when when then there is not gcode entered. Below a picture to show the result.
[image]

digant73, 2022-10-03:
@rondlh if you are familiar with git/PR, please submit you changes directly. remember to replace tabs with spaces and trim trailing spaces in the source code

rondlh, 2022-10-04:
> @rondlh if you are familiar with git/PR, please submit you changes directly. remember to replace tabs with spaces and trim trailing spaces in the source code. The same for your M300 FR. AFAIK, you already implemented it.

I will give it a try...

thisiskeithb, 2022-10-08:
> remember to replace tabs with spaces and trim trailing spaces in the source code.

@digant73 (and any developer contributing to this repo): Installing the [EditorConfig plugin for VSCode](https://marketplace.visualstudio.com/items?itemName=EditorConfig.EditorConfig) will use this repo’s [`.editorconfig`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/.editorconfig) file which will handle the defined formatting for you.

digant73, 2022-10-08:
@thisiskeithb yes, there are a lot of plugins doing a lot of stuff. This one is providing the basic formatting required in this project and it could make sense to provide this suggestion also in the README file

gordo3di, 2022-10-21:
> @digant73 (and any developer contributing to this repo): Installing the [EditorConfig plugin for VSCode](https://marketplace.visualstudio.com/items?itemName=EditorConfig.EditorConfig) will use this [...]

Interesting. I just installed it and never knew it was a thing. Thanks

kisslorand, 2022-10-28:
@rondlh
Check PR #2620

rondlh, 2022-11-01:
> @rondlh Check PR #2620

That works very well and looks good, well done, but I found that sometimes the "Prev" and "Next" button don't respond to short presses. Especially with the Next button it's easy to reproduce (send 2 codes like M119, M122, shortly press Next), I can see the screen responds to the input, but the command doesn't show afterwards.

Some general suggestions about the console.
1. In my view it would be more logic to make the button order: "Previous Clear Next"
2. You could even change the button label into the command that is hidden behind it, that would be helpful, so don't show "Prev", just show "G28".

kisslorand, 2022-11-01:
> That works very well and looks good, well done, but I found that sometimes the "Prev" and "Next" button don't respond to short presses. Especially with the Next button it's easy to reproduce (send 2 [...]

That's because the history buffer is for the last 3 distinct sent commands, after sending 2 commands and pressing "Next" you recall the 3rd saved command, which is empty. Check on the "Send" button the text, it will show "Next 3/3" meaning it recalls the the 3rd of the last 3 saved commands. It's not a big deal to implement skipping the empty slots but I do not see it as an issue worth taking up flash space for that.

> 
> Some general suggestions about the console.
> 
> 1. In my view it would be more logic to make the button order: "Previous Clear Next"

Give it another shot now that you know there are always 3 slots available.

> 2. You could even change the button label into the command that is hidden behind it, that would be helpful, so don't show "Prev", just show "G28".

That would be a bad idea in case of a long saved code.

Anyway, nothing's written is stone, can be changed/polished by anybody at anytime.

rondlh, 2022-11-02:
> That's because the history buffer is for the last 3 distinct sent commands, after sending 2 commands and pressing "Next" you recall the 3rd saved command, which is empty. Check on the "Send" button [...]

**You could cut down the command to a fixed max length. But either way it would only work with a history of 2 items**

> 
> Anyway, nothing's written is stone, can be changed/polished by anybody at anytime.

That explains a lot, thanks!
I found something that doesn't work as I would expect...
1. Go to the console, type M119 and Send, and go Back
2. Press Prev, M119 will popup and press Send again and go Back
3. Now if you press Prev again, you will _not_ get the M119

kisslorand, 2022-11-02:
> I found something that doesn't work as I would expect...
> 
> 1. Go to the console, type M119 and Send, and go Back
> 2. Press Prev, M119 will popup and press Send again and go Back
> 3. Now if you press Prev again, you will _not_ get the M119

Fixed it (sort of).
Whenever you return to the keyboard view "Prev" will select the last saved gcode.

There are still things left for enhancement but I do not think they worth it, it's not an essential feature of the TFT.

rondlh, 2022-11-03:
> Fixed it (sort of). Whenever you return to the keyboard view "Prev" will select the last saved gcode.
>
> There are still things left for enhancement but I do not think they worth it, it's not an [...]

Nice, that does the trick, great job again!

rondlh, 2022-11-03:
An excellent solution was provided, much better than I was hoping for, well done!

closed 2022-11-03 by rondlh

rondlh, 2022-11-04:
UPDATE: Still found some odd behavior.
For example enter and send 3 command:
Command1
Command2
Command3
Then press "Prev" twice to get Command2 again and send it.
What command would you expect to appear when you press "Prev" now?
I would expect Command2, but you actually get Command3

A solution is adding the 3rd line given below:

```
lastIndex = ~nowIndex;  // trigger text box redraw
saveEnabled = false;
saveGcodeIndex = (keyboardData->gcodeIndex+ 1) % MAX_GCODE_COUNT;
```

kisslorand, 2022-11-06:
> UPDATE: Still found some odd behavior. For example enter and send 3 command: Command1 Command2 Command3 Then press "Prev" twice to get Command2 again and send it. What command would you expect to appear when you press "Prev" now? I would expect Command2, but you actually get Command3

"Prev" and "Next" cycles through the SAVED gcodes, not the SENT ones. As I said, I made it such that after sending a command, the index for the saved commands is reset in way that "Prev" will select the last SAVED command. In the example above Command3 was the last saved one that's why it's selected when pressing "Prev".
This can be changed to preserve the index on last used command from history but anyhow, after selecting SAVED Command2 and sending it, the previous SAVED command would be Command1, in any case it would not be Command2 again (the previous command of Command2 is Command1).

> A solution is adding the 3rd line given below:
> 
> ```
> lastIndex = ~nowIndex;  // trigger text box redraw
> saveEnabled = false;
> saveGcodeIndex = (keyboardData->gcodeIndex+ 1) % MAX_GCODE_COUNT;
> ```

This would break the chain of saving commands. As it is now, the first in line SAVED command is overwritten with the last distinct SENT command. Your proposal would make saving it in a random slot.

I am about to push a new commit to PR #2620 with a slight flash size reduction, so please let me know how would you prefer it, reset the index to the last saved in history or preserve it. (preserving the index would use more flash than resetting it to the last saved one)

PR #2620 (2022-11-10): Terminal rework
### Requirements

BTT or MKS TFT

### Description

This PR is a bugfix and an enhancement with the added benefit of firmware flash size reduction.

- fixes a bug where in terminal keyboard if a sent gcode was recalled and sent, it was saved again in a new slot, hence having the same gcode saved in more or all slots available
- gcode sent history recall buttons visible while gcode text area is empty (no input) (credit goes to @rondlh, it is his idea #2609)

### Benefits

People using the terminal menu

### Related Issues

 As said above, in terminal keyboard if a sent gcode was  [... cut]