bigtreetech/BIGTREETECH-TouchScreenFirmware #2615: [BUG] TFT70 bmps are sized wrong. Version info is useless

thehans, 2022-10-19:
### Description

I tried updating my firmware, bmps, fonts, and config.ini on a BIQU BX, using the latest files from this repository
Firmware file: `BIGTREE_TFT70_V3.0.27.x.bin`
  
Now the icons on the screen are undersized, and the area that they don't cover leaves garbled image from previous screen.

Also the information screen just says:
`TFT70_V3.0.SOFTWARE_VERSION Mar 19 2021`
Same thing before and after the upgrade, so I don't know if it flashed newer firmware or not.

### Steps to reproduce

Compare these two files:
[001_mm.bmp from this TouchScreenFirmware repository](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_Unified%20Menu%20Material%20theme/TFT70/bmp/001_mm.bmp)
[001_mm.bmp from BIQU-BX repository](https://github.com/bigtreetech/BIQU-BX/blob/master/Firmware/Copy%20to%20microSD%20to%20update/TFT70/bmp/001_mm.bmp) (correctly sized).

**Expected behavior**
Correctly sized icons

Also Info screen `SOFTWARE_VERSION` should say `27` with a date that matches the github file-modified date.

**Actual behavior**
undersized icons

### Hardware Variant
TFT70 V3.0

### Additional Information
[image]
[image]

radek8, 2022-10-19:
You used the old FW. You downloaded the wrong icons..

radek8, 2022-10-19:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_Unified%20Menu%20Material%20theme/TFT70

digant73, 2022-10-19:
The TFT hw on Biqu BX printer has no MCU (a porting of the old TFT fw was integrated in Marlin fw 2.0.6.1. It is Marlin controlling the TFT). So you cannot use the TFT fw provided here on the BX. You need to reinstall Marlin 2.0.6,.1 fw.

thehans, 2022-10-19:
OK, I will try going back to factory firmware.  

It is still very misleading that the info screen claims to have a board of a regular TFT70 V3.0.  But now I see the PCB silkscreen says BIGTREETECH TFT70-BX V2.2.
So it is a screen specific to only this printer and there is only one firmware version that is 2 yrs old and I guess no plans for updates  :-/

closed 2022-10-20 by thehans