bigtreetech/BIGTREETECH-TouchScreenFirmware #2616: [BUG] TFT problem with Onboard SD card reading

NOVAXIM, 2022-10-23:
### Description
I just installed the latest version. However, I did notice some inconveniences with the last version which I used (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/186eb548e617dba5f5ff9817cd3e0cb15e8bc017).
The problem is that I cannot read the SD card from my motherboard when I have the old WiFi setup set to 115200 bouds. A window pops up with the title "Error:", no description and just OK. However, reading through the http console with the use of M21 and M20 L works great at 115200 bouds.
So I thought to reduce the WiFi transfer to 57200 and suddenly the SD card on the motherboard started working. However, not all is well. I noticed that I don't have one directory and sometimes some name (directory or file) is shown as not long.

### Steps to reproduce

As soon as I open the Marlin Mode and Touch Screen selection screen (and I don't have to go into Marlin) and select Touch Screen, then I already have all the directories, but the files are with short names.

**Expected behavior**

Can someone fix it?

**Actual behavior**

It works surprisingly strange. It shouldn't have to do with that.

### Hardware Variant

BIGTREETECH SKR MINI MZ V1.0 + TFT35 MZ V3.0

### TFT Firmware Version & Main Board Firmware details

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/73cf84f72995daf02d060bcb286bae06bd9a4ec5
https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.1

### Additional Information
When I was uploading the software, I wasn't using the reset.txt file, could that have a reason? However, I did a reset from the menu multiple times after that.

kisslorand, 2022-10-23:
From your description I assume you are talking about accessing the onboard SD from a PC via a browser and the issues you have are on the screen of your PC. If so, how is this TFT related?

NOVAXIM, 2022-10-23:
> From your description I assume you are talking about accessing the onboard SD from a PC via a browser and the issues you have are on the screen of your PC. If so, how is this TFT related?

You assumed wrong!
Please don't spoil the thread.

NOVAXIM, 2022-10-23:
When I set WiFi connection as OFF, then I can't read main board SD card. A window pops up with the title "Error:", no description and just OK. What related WiFi setting and reading SD card from main board (BIGTREETECH SKR MINI MZ V1.0) via TFT35 with Touch Screen mode?

kisslorand, 2022-10-23:
> Please don't spoil the thread.

Sure, carry on!

radek8, 2022-10-24:
@NOVAXIM 
If you have a problem with FW for TFT, only a programmer who works on that project can help you. 
But there are not many of them and they could be counted on the fingers of one hand.
And you just offended one of them

NOVAXIM, 2022-10-27:
@radek8
You must know that being offended is the result of mental problems. I don't suppose @kisslorand has anything with his head. Most likely he did not understand something, English is not my native language. 

It is worse with you, @radek8, because you are trying to offend me, which cannot be done. You are wrong that only a few people can program. I can fix it too (almost done). I am troubled, that you answer to my issues almost every time, but it is not substantively related to the topic. When you stop commenting and focus on the problem, it'll be nice. For me, this is just a place to resolve a problem, not a dispute with you. End & full stop.

NOVAXIM, 2022-10-27:
Dear contributors, I haven't worked much with this firmware and therefore I have a problem finding the right code location for a certain piece to analyze it.
The second problem is that I have Anycubic Mega Zero and it is very difficult to access the SD connector on the display board. I have to disassemble the entire printer every time. I know I can improve this by printing a new case, I don't have time for this. I only have a few hours a week for this.

Now, I can set any WiFi baudrate and onboard SD card working well. There is one strange activity that I am currently researching. I need check a new code and configurations effect. More time...

The above problem has two (if not three) causes. I'm well on my way to getting it out of the way and will be back here when I'm sure it's not a hardware defect. Now I close the issue.

ebraiman, 2022-11-15:
Here is a possible configuration to solve issue within Marlin v2.1.1  

https://www.youtube.com/watch?v=xK8F40LGnEE&list=PLP-VDZmS6bFopvc_qogCB7ompSvhqb3sm&index=19

closed 2023-02-05 by stale[bot]