bigtreetech/BIGTREETECH-TouchScreenFirmware #2618: Execute command for Nextion with M118 gcode.

s-safak, 2022-10-24:
Hi,
I am trying to control Nextion device with Skr Pro v1.2 But I couldn't find a way. 

I enabled  #define NEXTION_TFT nextion module and connected to TFT port. I used many versions of M118 codes but it didn't work.

Fo ex: M118 P2 page page2 or M118 A1 P2 page page2 or M118 P2 page page2 M118 P2 0xFF 0xFF 0xFF or whatever else....

These are serial settings. 
#define SERIAL_PORT -1
#define BAUDRATE 250000

#define SERIAL_PORT_2 1
#define BAUDRATE_2 9600   // Enable to override BAUDRATE

With this way, i can control SKr Pro with the Nextion device. For ex : print "M104 S210 T0"  it is working properly.

Now I need a hand to control pages or values on Nextion device with M118 command or another way.

Thank you from now

radek8, 2022-10-24:
How does the NEXTION display relate to this FW?

rondlh, 2022-11-04:
@s-safak Perhaps you can use the custom defined macros?
Could you be a bit more specific about what you want to do actually? With examples please...

avidNewb, 2022-11-21:
I'm wondering if this issue is related to a problem I am having.  I recently updated to the code in the master branch and commands I send to ESP3D don't work anymore.

Example:
Custom command in config:  M118 [ESP201] P23 V254 ANALOG=YES
Expected behavior: Set pin on ESP3D device to control light
Result:  Nothing happens

The same issue happens with commands in print files.  Previously I had notifications coded in my gcode files and they would be sent to the ESP3D.  None of them work anymore.  I have full control of the printer and tft from the ESP3D.

digant73, 2023-01-22:
@avidNewb fixed by #2678

closed 2023-02-15 by bigtreetech