bigtreetech/BIGTREETECH-TouchScreenFirmware #2619: BTT TFT35 V3.0.1 at BTT SKR2_RevB but the TFT shows "No printer attached"

MattenMax, 2022-10-26:
I am struggeling during upgrading my UM-Clone with the above mentioned board and display.
The Marlin firmware is 2.1.1, the TFT firmware is uptodate (BIGTREE_GD_TFT35_V3.0.27.x.bin). The board is correct connected to the TFT.
I have spent several days with reading in communities and watched several videos on the internet. But nothing helped so far...
Could anyone have a look into my settings, maybe there is somethin I missed...

Thank you so much!
Matten

Addes files: configuration.h, configuration_adv.h and config.ini for display

MattenMax, 2022-10-26:
[configs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/9872769/configs.zip)

kisslorand, 2022-10-26:
Test in "Configuration.h" with "#define SERIAL_PORT_2 1".

MattenMax, 2022-10-26:
Thanks for replying. I have tested these ports combinations:
-1/0, -1/1, -1/2, 0/1, 0/2.
Nothing worked.

gordo3di, 2022-10-26:
Make sure you followed everything here:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md#marlin-dependencies

kisslorand, 2022-10-26:
> Thanks for replying. I have tested these ports combinations: -1/0, -1/1, -1/2, 0/1, 0/2. Baud : 115200 Nothing worked.

Than check if RX & TX wires are swapped.

MattenMax, 2022-10-26:
Thanks a lot! 
I will definitely test it all out tomorrow.

MattenMax, 2022-10-26:
> Make sure you followed everything here:
> 
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md#marlin-dependencies

I just followed the instructions.... nothing has changed.

MattenMax, 2022-10-26:
> > Thanks for replying. I have tested these ports combinations: -1/0, -1/1, -1/2, 0/1, 0/2. Baud : 115200 Nothing worked.
> 
> Than check if RX & TX wires are swapped.

RX and TX? 
RX is connected to PA9 
TX is connected to PA10
is this correct?

kisslorand, 2022-10-27:
Yes, correct.
The correct wiring, configuring the serial ports and their baud rates are the only prerequisites to have a communication. You must have a hardware fault (TFT on mainboard).

MattenMax, 2022-10-27:
Good and clear summary. Thanks a lot. First I will check the wiring itself, after that I will send back the TFT...

radek8, 2022-10-27:
You have EEPROM disabled in your configuration
#define EEPROM_SETTINGS
Why?

radek8, 2022-10-27:
> > Make sure you followed everything here:
> > https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md#marlin-dependencies
> 
> I just followed the instructions.... nothing has changed.

You searched wrong

radek8, 2022-10-27:
Send photos of wiring and current configuration.

radek8, 2022-10-27:
If you have defective HW and you have the option to return the defective piece to the seller, then do it.
But you have to know if it is suitable for SKR or TFT.
If you can't return the defective piece, you can try to remap the defective others on the TX and RX pins.
It has already been addressed in several discussions here.

MattenMax, 2022-10-27:
I have just posted the current configuration after adjusting.
[configs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/9880628/configs.zip)
here are some photos, I hope you can see the relevant things
[image]
[image]

kisslorand, 2022-10-27:
> The board is correct connected to the TFT.

> RX is connected to PA9
> TX is connected to PA10

We were giving you hints based on wrong information provided. Anyway, glad you find the issue. Happy printing!

MattenMax, 2022-10-27:
Yeah I am very sorry. I did really double check it before posting here... But for some reason...

closed 2022-10-27 by MattenMax

radek8, 2022-10-28:
You are not the first to confuse the SWD and TFT connectors :-)