bigtreetech/BIGTREETECH-TouchScreenFirmware #2622: [FR] Movement speed control in the movement menu

rondlh, 2022-11-01:
I often need to control the movement speed in the movement menu to test my printer (test resonance frequencies and max speeds etc). It would be great to have this capability in the movement menu directly. Now I need to revert back to the console to handle this.

Possible solutions:
Option 1. Use the control knob push down function to popup a movement speed menu.
Option 2. Instead of using the turning of the knob to make a move, use the knob to set the movement speed directly (perhaps configurable by the user if they want the knob to make a move or to set the movement speed).

oldman4U, 2022-11-01:
kisslorand is waiting for your answer in the other ticket

rondlh, 2022-11-01:
> kisslorand is waiting for your answer in the other ticket
Thanks for the reminder... I replied just now.

closed 2023-01-10 by stale[bot]