bigtreetech/BIGTREETECH-TouchScreenFirmware #2623: MacOS 13 Ventura hidden files listed on TFT screen

slwise, 2022-11-01:
Since upgrading to MacOS 13 Ventura the hidden files are now being displayed with the gcode files when selecting the SD card on the TFT35. My friend also upgraded to Ventura and is experiencing the same issue. He placed a SD card created with Monterey and the hidden files were not displayed. He used a SD card created with Ventura and the hidden files appeared in the list. We are both experiencing the same issue since the upgrade. He is running the prior TFT35 firmware version and I am running the latest. I do not know of any way of preventing the MacOS from writing these files to the SD card. It appears the Mac has always been writing these files but until Ventura they never showed up in the SD card list. The TFT35 still functions correctly but this is more an annoyance because of the extra clutter in the list especially if you have a lot of files on the SD card.

[image]
[image]

oldman4U, 2022-11-02:
Have you formatted the SD card using macOS disk util or a 3rd party software and which file format do you use?

slwise, 2022-11-02:
I use the macOS disk utility and format them as FAT32. They are 8GB cards and I have 5 I use in rotation. I've been doing it this way for several years with no problems until Ventura. My friend has pretty much the same set up as I do and he is experiencing the same issue. The SD cards hadn't been formatted since I had the macOS Monterey so I reformatted then with Ventura but it didn't help. The hidden files have always been there even before Ventura but for some reason the TFT is now displaying them. Thanks for your help.

oldman4U, 2022-11-02:
Try this one, maybe it still works: https://www.sdcard.org/downloads/formatter/

As you know, macOS and FAT are not best friends.

Ventura is the problem in this case;-(

slwise, 2022-11-02:
I tried the SD formatter you suggested using the full format and the issue was still there. It was worth the try. I appreciate your help. Maybe after a Ventura update or two maybe it will fix itself.

oldman4U, 2022-11-02:
Hopefully. I always have a second volume with an older version so in case the printer driver or something similar does not work.

closed 2023-01-10 by stale[bot]