bigtreetech/BIGTREETECH-TouchScreenFirmware #2624: Cold extrusion prevented

Sergeykurm, 2022-11-02:
TFT50 GD v3.0.1

Printing from SD card MB SKR 3 EZ does not start. (Marlin 2.1.1)
Writes cold extrusion prevented.

Print start:

- 

echo:Now fresh file: /2.GCO
File opened: 2.GCO Size: 39565
File selected
//action:notification 2.GCO
ok
//action:resume
//action:prompt_end
//action:prompt_begin Resuming SD
//action:prompt_button Dismiss
//action:prompt_show
//action:notification 2.GCO
ok
//action:notification Bed Heating...
Writing to file: /PLR
  T:21.20 /0.00 (3936.0000) B:21.37 /0.00 (3931.2500) @:0 B@:0
echo: cold extrusion prevented
echo:Power Outage
//action:notification 2.GCO
//action:notification Power Outage:
Error:Printer halted. kill() called!
//action:poweroff

- 

Somewhere I saw something similar. Nothing helped me. Can you tell me what to do? How to fix?

kisslorand, 2022-11-02:
What if you print from SD card?

Sergeykurm, 2022-11-03:
When I print from an SD card, it says "Power Outage".

Above is an example of printing from an SD card via ESP3D.

kisslorand, 2022-11-03:
I might be wrong but it seems to me you might have the nozzle heater wires shorted.

Sergeykurm, 2022-11-03:
**kisslorand**

It prints great from the SD screen.

kisslorand, 2022-11-03:
Post a gcode please.

Sergeykurm, 2022-11-03:
**kisslorand**
Gcode.
I don't know how to embed the file. Here is the beginning of Gcode.

- 

;FLAVOR:Marlin
;TIME:1106
;Filament used: 0.432071m
;Layer height: 0.2
;MINX:102.9
;MINY:102.9
;MINZ:0.2
;MAXX:122.1
;MAXY:122.1
;MAXZ:20
;LINEARADVANCEPROCESSED
;Generated with Cura_SteamEngine main
M140 S85
M105
M190 S85
M104 S227
M105
M109 S227
M82 ;absolute extrusion mode
M300 P250 S300 ; Звук

; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G92 E0
G92 E0
G1 F1500 E-1
;LAYER_COUNT:100
M900 K0.180000 T0 ;added by LinearAdvanceSettingPlugin
;LAYER:0
M107
M204 S500
M205 X8 Y8
G0 F6000 X104.339 Y104.771 Z0.2
G0 X102.901 Y109.098
M204 S1200
;TYPE:SKIRT
G1 F1500 E0
G1 F900 X102.91 Y108.433 E0.02212
G1 X102.959 Y107.766 E0.04436
G1 X103.087 Y107.109 E0.06663
G1 X103.292 Y106.473 E0.08885
G1 X103.572 Y105.865 E0.11112
G1 X103.923 Y105.295 E0.13338
G1 X104.339 Y104.771 E0.15563
 .......and so on
-

Sergeykurm, 2022-11-03:
Gcode any does not start. On the Li ear advance website, the test one also does not work. I think it's not about gcode. It seems to me that you need to look for a screen or MB firmware error.

radek8, 2022-11-03:
Does it behave the same when you start printing from the onboard SD in touch mode and when you start printing from Merlin simulated mode?

Sergeykurm, 2022-11-03:
> Does it behave the same when you start printing from the onboard SD in touch mode and when you start printing from Merlin simulated mode?

From the simulated Merlin mode I haven't tried it. Haven't figured it out yet. There I can not find "Start Printing" and "Select SD and Files".

radek8, 2022-11-03:
Do you have SD card printing enabled in Merlin? That could be causing your problem.

Sergeykurm, 2022-11-03:
#define SDSUPPORT

radek8, 2022-11-03:
first to comment:
 //#define REPRAP_DISCOUNT_SMART_CONTROLLER
second to uncoment:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

in TFT
enable 128x64 display on the display

set config.ini fot TFT
marlin_type:0

radek8, 2022-11-03:
Be careful not to confuse you.
To which board and with which cable will you connect TFT and SKR?
will you use two cables EXP1 and EXP2, or one cable EXP3?
There is a different display setting in Merlin for one EXP3 cable.

Sergeykurm, 2022-11-03:
All right. Everything is set up correctly.

Sergeykurm, 2022-11-03:
EXP1 and EXP2   // on

Sergeykurm, 2022-11-03:
Looks like I found the cause of the cold extruder. Now START PRINTING from the SKR SD card.

The reason was in the line:
#define POWER_LOSS_PULLUP
I commented it out and it worked.

I don't have UPS installed. Apparently at the beginning of printing there was a conflict between UPS and emergency recording on the SD card.
  #define PLR_ENABLED_DEFAULT true.
PS
Thanks to everyone who responded. I will continue to look for errors, I want to configure Marlin 2.1.1 so that everything works perfectly.

Sergeykurm, 2022-11-03:
Закрываю.

closed 2022-11-03 by Sergeykurm