bigtreetech/BIGTREETECH-TouchScreenFirmware #2631: Show print screen / progress bar while printing

tom-321, 2022-11-11:
**Your question or details you need. Please describe.**
No idea if this is a matter of the display, but I don't know what else to set in Marlin (I've uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) 

I'm trying to show the print screen
I print over USB (Repetiere Server) and the TFT35 V2.0 ist connected to the skr 2 rev b (Marlin 2.1.1) over TFT display pins / serial port.
Serial port 1 and 2 is activated in firmware

What do I have to do to display the print screen which shows the progress ?

radek8, 2022-11-12:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2180

radek8, 2022-11-12:
Unfortunately, your display does not have a simulated Merlin display mode.

rondlh, 2022-11-13:
@tom-321 What you want is possible, but is a use case that is not in the current scope of the software provided here. You would need some customization to achieve this. Marlin can generate the required codes (SET_PROGRESS_MANUALLY, M73), the TFT needs to process these commands and show them. Not a big deal, but outside of the scope here.

tom-321, 2022-11-19:
@rondlh Thanks for the advice
If I have understood this correctly M73 is not in the gcode, ist generated by marlin to set displayed progress of the connected serial devices 

But what I don't understand is why it doesn't work then.
I have defined LCD_SET_PROGRESS_MANUALLY in the configuration_adv. 
On my Ender3 v2 (with octoprint) it also works fine...

rubienr, 2022-11-26:
Hi @tom-321,

I have seen many issues reporting `M117` and/or `M73` not working properly. What I found by scrolling the source is that:

1. the print screen on the BTT TFT device is not activated because it expects to see `// action:print_start` and that action isn't sent by Marlin nor OctoPrint.
2. from my perspective [`M73`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L794) is not working as expected or documentation of `PROG_SOURCE` in [Configuration.h#L256](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h#L256)  might not be correct/complete.
3. OctoPrint does not send the expected actions (`// action:command`) for start, stop, pause, resume and cancel.
4. Cura does not provide the expected strings (see [parseACK.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/parseACK.c))
5. it works with [`M117`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L908) followed by: `Time Left %dh%dm%ds` or `Layer Left %d/%d` or `Data Left %d/%d`. 

**Workaround for 1 and 3 to activate the print screen:**

In OctoPrint go to settings -> Printer -> G-Code Scripts and enter those respectively:

  - `M118 A1 P2 action:print_start`
  - `M118 A1 P2 action:print_end`
  - `M118 A1 P2 action:cancel`
  - `M118 A1 P2 action:pause`
  - `M118 A1 P2 action:resume`
 
Note: you might need to change `Pn` (here `P2`) to something else.

**Workaround for 2 and 4 to populate remaining time, progress and layer to the display:**

I couldn't get it running with `M73 P%d`. My naive hotfix is to add layer, time and progress information by Cura  and then sring-replace those with `M117 ...`.

In Cura -> Extensions -> Post Processing -> Modify G-Code:

1. add "Diasplay Filename And Layer On LCD": scroll-enabled false; text to display empty; initial layer number 0; display max layer enabled
2. add "Search and Replace": search `M117\s*Layer\s*(\d*)\s*of\s*(\d*)\s.*`; replace `M117 Layer Left \1/\2`; use regexp enabled
3. add "Display Progress on LCD": time remaining enabled; time reporting method M117; update frequency every layer; percentage enabled
4. add "Search and Replace": search `M73\s*P(\d*).*`; replace `M117 Data Left \1/100`; use regexp enabled

This adds lines in the G-Code i.e. as 
```gcode
M117 Time Left 0h00m34s
M73 P50
M117 Layer 30 of 60 somefilename
```
and replaces to
```gcode
M117 Time Left 0h00m34s
M117 Data Left 50/100
M117 Layer Left 30/60
```

Bug investigation:

I can see that in [interfaceCmd.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L908) `M17` works as expected
```C++
        case 117:  // M117
          if (cmd_seen_from(cmd_base_index, "Time Left"))  // parsing printing time left
          {
            // format: Time Left <XX>h<YY>m<ZZ>s (e.g. Time Left 02h04m06s)
            parsePrintRemainingTime(&cmd_ptr[cmd_index]);  // cmd_index was set by cmd_seen_from function
            setTimeFromSlicer(true);
          }
```
but `M73` in [interfaceCmd.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L764) not
```C++
        case 73:
          if (cmd_seen('P'))
          {
            setPrintProgressSource(PROG_SLICER);
            setPrintProgressPercentage(cmd_value());
          }
```
With settings and Features:
- [`#define PROG_SOURCE 1`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h#L256)
- [`#define SET_PROGRESS_MANUALLY 
// Add the G-code 'M73' to set / report the current job progress`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h#L1473)
- [`#define M73_REPORT // Report progress to host with 'M73'`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h#L1481)

**How to reproduce:**

In OctoPrint console send:

```gcode
M118 A1 P2 action:print_start
M117 Time Left 0h00m34s
; set progress
M117 Data Left 50/100
M117 Layer Left 30/60
```

```gcode
M118 A1 P2 action:print_start
; set progress doesn't work
M73 P50
```

tom-321, 2022-12-01:
[image]

rubienr, 2022-12-04:
In my opinion, 

- something goes wrong with [`M73`](https://marlinfw.org/docs/gcode/M073.html) (even if sent directly to the display's serial port by `M118 A1 P2 M73 ...`), and
- the command `M117 (Layer|Time|Data) Left \d*/\d*` is hijacked if it is provided in the afore mentioned form but that fact is not well documented and the specific behaviour cannot be disabled. 

My experience with the print screen is ambivalent. While the upper screen part providies a very nice and well done  progress overview, the lower buttons are somehow limiting. I wish the print screen wouldn't stay sticky in the foreground, allow me to go the the status screen (i.e. "More" button cold be the back-arrow) and back again (from the main screen).

Another issue I observed while "Remote printing..." is shown on top of the screen:  it does not allow me to successfully use the pause/resume and abort buttons because it says "Busy processing, please wait..." .

My current workaround is:

1. not sending `M118 A1 P2 action:print_start` on print start
2. replace the move/homing menu item on main screen (2nd button) with the printing menu: [MainPage.c](https://github.com/rubienr/btttft/commit/ede20e34c5f1ff49552fae64b960da3d192c66cc#diff-31283603695fa04268df00b4b912278ae48172071e5f9081e66522f79bd94aa5)
3. replace the print menu item (bottm right button) on status screen with move/homing: [StatusScreen.c](https://github.com/rubienr/btttft/commit/f60bd760ec8ce0da7b5d9849e9081131441ca5b3#diff-d13705e9c4855499a327156733e9367921ca686bae6f53b0fcbd90de1585a6a0)
4. activate the printing screen manually from the main screen (drawback: some counters do not auto update since the firmware never detected the print start)

As I don't need the SD-card support (and its respective menu) for me this is sufficient at the moment.

digant73, 2023-01-26:
@rubienr, @tom-321 As far as I understood, only M73 is not properly working (I'm 100% sure it was working in the past) even if you print from TFT and also remotely (e.g. Octoprint connected to one TFT's serial port), Right?
IMHO M117 Layer/Time etc... are properly documented. I confirm that the text in M117 is case sensitive (as any other matching key provided in parseAck.c. E.g. M117 Time Left xxxx is the one to use while M117 time left xxxx (lower case) is wrong.

digant73, 2023-01-26:
> Hi @tom-321,
> 
> I have seen many issues reporting `M117` and/or `M73` not working properly. What I found by scrolling the source is that:
> 
> 1. the print screen on the BTT TFT device is not activated because it expects to see `// action:print_start` and that action isn't sent by Marlin nor OctoPrint (I don't know how Repetier behaves).
> 2. OctoPrint does not send the expected actions (`// action:command`) for start, stop, pause, resume and cancel.
> 3. from my perspective [`M73`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L794) is not working as expected or documentation of [`PROG_SOURCE`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h#L256)  might not be correct/complete.
> 4. it works with [`M117`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L908) followed by: `Time Left %dh%dm%ds` or `Layer Left %d/%d` or `Data Left %d/%d`.
> 5. Cura does not provide the expected strings for `M117` (see [parseACK.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/parseACK.c))
> 
> **Workaround for 1 and 2 to (de-)activate the print screen:**
> 
> In OctoPrint go to Settings → Printer → G-Code Scripts and enter those respectively:
> 
> * `M118 A1 P2 action:print_start`
> * `M118 A1 P2 action:print_end`
> * `M118 A1 P2 action:cancel`
> * `M118 A1 P2 action:pause`
> * `M118 A1 P2 action:resume`
> 
> Note: you might need to change `Pn` (here `P2`) to something else.
> 
> **Workaround for 4, 5 to populate remaining time, progress and layer to the display:**
> 
> I couldn't get it running with `M73 P%d`. My naive hotfix is to add layer, time and progress information to the G-Code by Cura and then sring-replace those with `M117 ...` commands. Unfortunately the documentation lacks and its not obvious that commands of pattern `M117 (Layer|Time|Data) Left \d*/\d*` are hijacked and not printed to display as `M117` should do. Example: `M117 layer Left xxx` would go to display, whereas in contrary `M117 Layer Left xxx` would be hijacked, parsed but not be printed at all.
> 
> In Cura → Extensions → Post Processing → Modify G-Code:
> 
> 1. add "Display Filename And Layer On LCD": scroll-enabled false; text to display empty; initial layer number 0; display max layer enabled
> 2. add "Search and Replace": search `M117\s*Layer\s*(\d*)\s*of\s*(\d*)\s.*`; replace `M117 Layer Left \1/\2`; use regexp enabled
> 3. add "Display Progress on LCD": time remaining enabled; time reporting method M117; update frequency every layer; percentage enabled
> 4. add "Search and Replace": search `M73\s*P(\d*).*`; replace `M117 Data Left \1/100`; use regexp enabled
> 
> This adds lines in the G-Code i.e. as
> 
> ```gcode
> M117 Time Left 0h00m34s
> M73 P50
> M117 Layer 30 of 60 somefilename
> ```
> 
> and replaces to
> 
> ```gcode
> M117 Time Left 0h00m34s
> M117 Data Left 50/100
> M117 Layer Left 30/60
> ```
> 
> **Bug investigation:**
> 
> I can see that in [interfaceCmd.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L908) `M117` works as expected
> 
> ```c++
>         case 117:  // M117
>           if (cmd_seen_from(cmd_base_index, "Time Left"))  // parsing printing time left
>           {
>             // format: Time Left <XX>h<YY>m<ZZ>s (e.g. Time Left 02h04m06s)
>             parsePrintRemainingTime(&cmd_ptr[cmd_index]);  // cmd_index was set by cmd_seen_from function
>             setTimeFromSlicer(true);
>           }
> ```
> 
> but `M73` in [interfaceCmd.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L794) not
> 
> ```c++
>         case 73:
>           if (cmd_seen('P'))
>           {
>             setPrintProgressSource(PROG_SLICER);
>             setPrintProgressPercentage(cmd_value());
>           }
> ```
> 
> With:
> 
> * Marlin [bugfix-2.1.0](https://github.com/MarlinFirmware/Marlin/commit/8298a477e1f1fa6188788448eec3f349c1ba93de)
>   
>   * [`#define SET_PROGRESS_MANUALLY  // Add the G-code 'M73' to set / report the current job progress`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h#L1473)
>   * [`#define M73_REPORT // Report progress to host with 'M73'`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h#L1481)
> * Bigtreetech [TFT70_V3.0 FW V27.x](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/28cf2f4d68930e6482d20b9470e49048cc04f49e)
>   
>   * [`#define PROG_SOURCE 1`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h#L256)
> * Ultimaker Cura [5.2.1](https://github.com/Ultimaker/Cura/tree/5.2.1)
> * OctoPrint [1.8.6](https://github.com/OctoPrint/OctoPrint/tree/1.8.6)
> 
> **How to reproduce:**
> 
> In OctoPrint console send:
> 
> ```gcode
> M118 A1 P2 action:print_start
> M117 Time Left 0h00m34s
> ; set progress
> M117 Data Left 50/100
> M117 Layer Left 30/60
> ```
I confirm this is a correct flow that should properly work
> 
> ```gcode
> M118 A1 P2 action:print_start
> ; set progress doesn't work
> M73 P50
> M118 P2 M73 P50
> ```
With the exception of `M118 P2 M73 P50` (wrong command not supported by TFT), even this flow should properly work. M73 should be properly parsed and progress properly display on Printing menu
> 
> Note: with btttft in Marlin mode `M73` works as expected.

digant73, 2023-01-29:
> In my opinion,
> 
> * something goes wrong with [`M73`](https://marlinfw.org/docs/gcode/M073.html) (even if sent directly to the display's serial port by `M118 A1 P2 M73 ...`), and
> * the command `M117 (Layer|Time|Data) Left \d*/\d*` is hijacked if it is provided in the afore mentioned form but that fact is not well documented and the specific behaviour cannot be disabled.
> 
> My experience with the print screen is ambivalent. While the upper screen part providies a very nice and well done progress overview, the lower buttons are somehow limiting. I wish the print screen wouldn't stay sticky in the foreground, allow me to go the the status screen (i.e. "More" button cold be the back-arrow) and back again (from the main screen).
> 
> Another issue I observed while "Remote printing..." is shown on top of the screen: it does not allow me to successfully use the pause/resume and abort buttons because it says "Busy processing, please wait..." .
> 
> My current workaround is:
> 
> 1. not sending `M118 A1 P2 action:print_start` on print start
> 2. replace the move/homing menu item on main screen (2nd button) with the printing menu: [MainPage.c](https://github.com/rubienr/btttft/commit/ede20e34c5f1ff49552fae64b960da3d192c66cc#diff-31283603695fa04268df00b4b912278ae48172071e5f9081e66522f79bd94aa5)
> 3. replace the print menu item (bottom right button) on status screen with move/homing: [StatusScreen.c](https://github.com/rubienr/btttft/commit/f60bd760ec8ce0da7b5d9849e9081131441ca5b3#diff-d13705e9c4855499a327156733e9367921ca686bae6f53b0fcbd90de1585a6a0)
> 4. activate the printing screen manually from the main screen (drawback: some counters do not auto update since the firmware never detected the print start)
> 
> As I don't need the SD-card support (and its respective menu) for me this is sufficient at the moment.
> 
> @tom-321 documentation related to print screen + remote printing: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#printing-from-remote-host

The Feature Request (FT) about availability of Pause/Resume and Abort on remote printing was implemented in #2678. I currently closed it until some constraints from another contributor are accepted or rejected by btt.

rubienr, 2023-02-03:
Hello @digant73 ,

> As far as I understood, your use case is printing from OctoPrint connected to the mainboard instead of connected to one of TFT's serial port), Right? I will have a look on the M73 issue you reported, if any

Yes: printing from OctoPrint via USB (serial) not from SD-card via BTT-TFT serial.

> **EDIT:** M73 is currently not supported only when printing from (remote) onboard media. In that case even if:
> 
>     * `SET_PROGRESS_MANUALLY`
> 
>     * `M73_REPORT`
> 
> 
> are enabled in Marlin, the notification: `echo: M73 Progress: <XX>%; Time left: <YY>m;` sent by Marlin (for each M73) is currently not parsed by the TFT fw. Furthermore, an annoying echo message is instead displayed. This bug has been fixed and it will be provided in a next PR.

:smiley_cat:  Thank you!

PR #2692 (2023-02-15): Bug fixes printing from (remote) onboard media
**BUGFIXES:**
* Fixed a bug introduced by #2599: The matching rule on parseAck.c became a sink once a print from (remote) onboard is started and remained a sink until the media is changed (e.g. a print from SD card is started). This bug did not allow the parsing of any command (e.g. sent by Terminal or remote host) whose matching rule followed the sink.
* Fixed missing M73 parsing when printing from (remote) onboard media reported in #2631

  Furthermore, in case:
  * `SET_PROGRESS_MANUALLY`
  * `M73_REPORT`

  were enabled in Marlin, an annoying  [... cut]
closed 2023-02-15 by bigtreetech