bigtreetech/BIGTREETECH-TouchScreenFirmware #2632: [FR] Input shaping setting under Machine settings.

MLiv79, 2022-11-12:
Marlin is adding input shaping (finally 👍 ) in their latest bugfix 2.1.x versions. Could you add the item in the machine settings? Thanks

radek8, 2022-11-12:
What is input shaping?

kisslorand, 2022-11-12:
https://github.com/MarlinFirmware/Marlin/pull/24797

kisslorand, 2022-11-12:
I wanted to add this to the TFT firmware but this feature is still infantile, it's in a very early stage. For me it's not even working at all so it might go over major overhauls until it will be a solid working one.
If others don't get ahead of me I will surely include it in the firmware of the TFT once it will be solid.

thisiskeithb, 2022-11-13:
Input Shaping / `M593` parameters aren't likely to change in the near future, but you can follow https://github.com/MarlinFirmware/Marlin/pull/24951 for updates until Marlin `2.1.2` is tagged & released.

### Current Parameters:
 *  `D<factor>`: Set the zeta/damping factor. If axes (X, Y, etc.) are not specified, set for all axes.
 *  `F<frequency>`: Set the frequency. If axes (X, Y, etc.) are not specified, set for all axes.
 *  `X<1>`: Set the given parameters only for the X axis.
 *  `Y<1>`: Set the given parameters only for the Y axis.

### Not implemented yet / ZV is the default:
 *  `T[map]`: Input Shaping type, 0:ZV, 1:EI, 2:2H EI

gumanachile, 2022-12-29:
The Input Shaping is official now, are you reconsidering to add it to the tft firmware? We will really appreciate it! I have a Sidewinder X2 and its very annoying to input data from terminal having this awesome tft firmware!

arminth, 2023-01-21:
I second that!

kisslorand, 2023-01-24:
I started working on it...

gumanachile, 2023-01-24:
But if it is in machine/settings it doesn’t need icons 👍🏽

kisslorand, 2023-01-24:
Great idea, thank you!

kisslorand, 2023-01-25:
Done #2689

maniac75, 2023-01-29:
> Done #2689

Great!! Can't wait until the release of the .bin!!
Thank you!

kisslorand, 2023-01-29:
No need to wait, I can compile it for you, just let me know what TFT do you have.

maniac75, 2023-01-29:
> No need to wait, I can compile it for you, just let me know what TFT do you have.

Thank you!! I have the tft28 for the artillery sidewinder x2 🙏🏽

kisslorand, 2023-01-29:
What bin have you used before to update your TFT? I am asking because I am not sure ASWX2's TFT is with STM32F107xx MCU, like the ASWX1's TFT.

maniac75, 2023-01-29:
> What bin have you used before to update your TFT? I am asking because I am not sure ASWX2's TFT is with STM32F107xx MCU, like the ASWX1's TFT.

[image]

maniac75, 2023-01-29:
> What bin have you used before to update your TFT? I am asking because I am not sure ASWX2's TFT is with STM32F107xx MCU, like the ASWX1's TFT.

Yes. Its stm32, the same like the x1

kisslorand, 2023-01-29:
Here it is: https://www.dropbox.com/t/pmJX1jtfN7E5yMV1

maniac75, 2023-01-29:
> Here it is: https://www.dropbox.com/t/pmJX1jtfN7E5yMV1

Thank you!! I will try it asap!!

LTSpenny, 2023-01-30:
any chance at getting it compiled for a TFT 43, using TFT43V3.0.27 APR10 2021?

kisslorand, 2023-01-30:
Sure, here you have it: https://www.dropbox.com/t/DeIqmdNXo1qq945h

maniac75, 2023-01-30:
[image]
[image]

Works great!! Thanks @kisslorand !!!

LTSpenny, 2023-02-01:
> Sure, here you have it: https://www.dropbox.com/t/DeIqmdNXo1qq945h

That file says its for tft28. Thatll work on a tft43? [image]

kisslorand, 2023-02-01:
Sorry, my mistake, here's the file for TFT43: https://www.dropbox.com/t/SOnXahZwtHl0VykN

LTSpenny, 2023-02-01:
> Sorry, my mistake, here's the file for TFT43: https://www.dropbox.com/t/SOnXahZwtHl0VykN

Thank you so much!

Djungelman, 2023-02-03:
Hello. Is It possible to have a compilation for BIGTREE_GD_TFT35_V3.0_E3.27? Thank you

kisslorand, 2023-02-03:
Sure: https://www.dropbox.com/t/8aGmFFugYFM17xHq

Djungelman, 2023-02-03:
Thank you very much. I have Input Shaping X and Y also Shaping_Meny enabled in Marlin and had no problem to install the files you sent. But there is no Input Shaping meny in Settings. Reg// Djungelman

kisslorand, 2023-02-03:
Did you install the language file also?

Djungelman, 2023-02-03:
Yes. Your .BIN and Laguage files, From Github: TFT 35 Unified, Config.ini. No problem to update the TFT what I can see.

kisslorand, 2023-02-03:
Update please with the language file from the archive. I included them for a reason.

Djungelman, 2023-02-03:
I Take a clean sd card, copy the folder TFT35 and Config.ini from latest Github file. Then I copy your .bin and the Language Pack I got from you. The TFT seems to update with no errors but no meny for Input Shaping will appear. If I goes to Terminal and type M593 it respond with: M593 X F40.00 D0.15 and M593 Y F40.00 D0.15.

kisslorand, 2023-02-03:
See above

Djungelman, 2023-02-03:
Yes, I am in the Menu->Setting->Machine->Settings. But no Input Shaping. I a couple printers with the newest Marlin with Input Shaping but with the BIGTREE_TFT35_V3.0_E3.27.x.bin. If you have the time to compile that .Bin I will try on the other printers. //Thank you for the time and help.

Djungelman, 2023-02-03:
I take  another TFT screen: TFT28 and tried to update with the files you sent to maniac75 . Same problem. No Input Shaping on that eighter. I must be an idiot. Cant figure out what I am doing wrong. Sorry for your time..

kisslorand, 2023-02-03:
Here are the bin files for all the TFT variants: https://www.dropbox.com/t/y5NMFrGTVkpSNUXL

maniac75, 2023-02-03:
> I take  another TFT screen: TFT28 and tried to update with the files you sent to maniac75 . Same problem. No Input Shaping on that eighter. I must be an idiot. Cant figure out what I am doing wrong. Sorry for your time..

Try to reset the tft with an empty file named reset.txt and copy it in a blank sd and reboot

PR #2689 (2023-02-09): Input Shaping
### Requirements

MKS or BTT TFT

### Description

This PR adds the option to be able to enter the "__input shaping__" parameters (if IS is enabled) from the TFT.

Hungarian language is updated with the new "input shaping" translation but it means "resonance attenuation" as "input shaping" translated to Hungarian has a very ambiguous meaning.

### Related Issues

Fixes #2632
closed 2023-02-09 by bigtreetech