bigtreetech/BIGTREETECH-TouchScreenFirmware #2638: [BUG] (short description)

avidNewb, 2022-11-22:
### Description

M118 Commands do not go through to ESP3D when using repo code (not release).  M118 commands work fine in all other scenarios

Example:
M118 P0 [ESP201] P23 V254 ANALOG=YES 
**Should be received by ESP3d and set pin 23 to full power**

Source:
ESP3D WebUI:  Passed
Onboard SD:  Passed
Motherboard custom commands: Passed
TFT SD:  Fail
TFT USB:  Fail
TFT Custom Commands:  Fail
TFT Terminal:  Cannot test due to no brackets in terminal

### Steps to reproduce

1. Compile and load master branch on TFT
2. Place gcode for M118 command in any of the failed locations mentioned above
3. Run gcode

**Expected behavior**
Command is sent to all serial ports and executes on ESP3D

**Actual behavior**
Command never reaches ESP3D.  No output

### Hardware Variant
BTT TFT35 E3 V3.0
### TFT Firmware Version & Main Board Firmware details
Master branch, Marlin 2.1.1, ESP3D 2.1 & 3.0

### Additional Information

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10065594/Configuration.zip)

avidNewb, 2022-11-22:
I pulled the code from commit 99e230a650267fd172c9aacfc59dcaeb2e6997bd simply because I wanted to retain the Event Led feature and M118 commands work perfectly.  So sometime between that commit and now the ability to send M118 commands was broken.

avidNewb, 2022-11-23:
So to version of the code I pulled does have M118 working correctly but otherwise doesn't work with Marlin 2.1.2.  I'm an now back on the latest master branch code build and M118 is not functioning.  I'm going to try a later commit that is compatible with Marling 2.1.2 and see what the results are

avidNewb, 2022-12-08:
> If ESP3D is plugged in the TFT then the behavior you reported is correct because when the M118 is generated by TFT (e.g. read from TFT's SD card) then the response from Marlin is not forwarded by the TFT to other devices that didn't originated the request. If the TFT receives the request from the ESP3D then the response is properly forwarded by the TFT to ESP3D (to the port the TFT received the request). Probably an old version of the TFT firmware was forwarding the Marlin's response to all the TFT's ports by mistake. I remember I fixed something like that in the past due to another bug report complaining of ESP3D was receiving responses for requests not originated by the ESP3D that were not properly handled by the ESP3D.
> 
> If ESP3D is plugged in the mainboard then the issue is on Marlin.

I don't believe that is correct.  M118 is used to send commands to all serial devices.  a "P" parameter is used to specify the serial port or if no port is specified then all ports.  In the scenario you propose, how can I ever get commands from a gcode file to the ESP3D connected to the TFT?

digant73, 2023-01-22:
@avidNewb fixed by #2678

avidNewb, 2023-02-05:
> @avidNewb fixed by #2678

All drama aside, is this pull request on track? Anything I can do to help?

digant73, 2023-02-06:
From my side, the more I can do is to reopen the PR and removing the so much important "stolen / borrowed" comments

closed 2023-02-15 by bigtreetech