bigtreetech/BIGTREETECH-TouchScreenFirmware #2639: BTT TFT 50 don´t show layer number and height

Mastalirsky1, 2022-11-22:
Hi,
I have question, I installed a new BTT SKR mini E3 V3 and BTT TFT 50 V3 on my Ender 3, built and modified firmware for mainboard and config file for the TFT with all needed settings which you have here, but it still not show current layer number and height when printing, do you have any advice how to make it work? Thank you very much.

kisslorand, 2022-11-23:
Where do you print from?

> it still not show current layer number and height when printing

What does it show?

Mastalirsky1, 2022-11-23:
I print from mainboard SD, status bar with percenetege, remaining time and all other info works fine, only in layer column is just ---

kisslorand, 2022-11-23:
What if you click/touch/press on the layer icon during print?

Mastalirsky1, 2022-11-23:
It only switch between layer number and layer height and under both it is showing only ---

rondlh, 2022-12-07:
Are the number of layers and current layer actually in the gcode you are printing?
Look for:
LAYER_COUNT:
LAYER:
What did you find?

kisslorand, 2022-12-07:
File comment parsing is not available when printing from mainboard SD.

rondlh, 2022-12-07:
> File comment parsing is not available when printing from mainboard SD.
Of course, so to get this to work you would need Marlin to parse the comments, and send the information to the TFT. You could use ExtUI(Marlin) for that, and pick up the info in parseAck.c(BTT TSFW). I use a similar approach to play M300 sounds on the TFT instead of via Marlin.

Mastalirsky1, 2022-12-08:
In gcode I have LAYER_COUNT: and LAYER:
So I have to print from TFT USB or what I have to do to make it work for mainboard SD too?

kisslorand, 2023-01-07:
>what I have to do to make it work for mainboard SD too?
 You would need to modify Marlin source code to parse those information and send it to the TFT

kisslorand, 2023-01-23:
@Mastalirsky1 
Do you have AUTO_REPORT_POSITION and EXTENDED_CAPABILITIES_REPORT enabled in Marlin (Configuration_adv.h)?

Mastalirsky1, 2023-01-29:
@kisslorand
Yes I do

kisslorand, 2023-01-29:
I am out of guess what could be the cause of your issue, I cannot replicate this on my machines.

closed 2023-04-25 by stale[bot]