bigtreetech/BIGTREETECH-TouchScreenFirmware #2640: [BUG] Screen freeze on onboard SD print abort

jcwtm, 2022-11-24:
### Description

The screen will be stuck at 'Loading...' or 'busy processing, please wait' when cancelling an onboard SD print. The message that shows depends on the gcode that is run. See below. This does not occur when printing from the TFT card slot.

### Steps to reproduce

1. Upload the following Gcode to an SD card:
G28
M109 S100
2. Insert the SD card into the motherboard.
3. Start the print.
4. Cancel the print during heating.
5. Press 'Back' button
6. Blank screen stuck showing 'Loading...' 

### Another way to reproduce

1. Upload the following Gcode:
G28
M140 S60
M105
M190 S60
M104 S100
M105
M109 S100
2. Insert the SD card into the motherboard.
3. Start the print.
3. Cancel the print during bed heating.
4. Screen freezes showing 'busy processing, please wait' 

**Expected behavior**
Run cancel gcode.

**Actual behavior**
Screen becomes frozen, does not run cancel gcode.

### Hardware Variant
B1 TFT35 V3.0.1
SKR MINI E3 V3.0
### TFT Firmware Version & Main Board Firmware details
Marlin Firmware: 2.1.1
TFT Firmware: BIGTREETECH_GD_TFT35_B1_v3.0 V27.x Nov 10 2022

### Additional Information
[bug.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10079953/bug.zip)

kisslorand, 2022-11-24:
The TFT will break the waiting for bed heating (M108) and send M524 (Abort SD print).
In Marlin 2.1.1 M524 is not part of the emergency commands so it will skip to the M140 command from the buffer and will block communication until the nozzle is heated. That's why your TFT remains in  'busy processing, please wait' state, it waits for Marlin to respond.

However in Marlin 2.1.2 M524 is part of the emergency commands family but the TFT will still freeze because of a bug. This bug is already fixed in PR #2607 but Bigtreetech is not merging it for 2 months now...
Solution to your problem is upgrading to Marlin 2.1.2 and use the forementioned PR. If needed, I can compile TFT FW for you with PR #2607 included.

jcwtm, 2022-11-25:
@kisslorand Thank you for the information.

kisslorand, 2023-02-05:
It is stale because the fix wasn't merged. The fix is in PR #2607.

closed 2023-02-09 by bigtreetech