bigtreetech/BIGTREETECH-TouchScreenFirmware #2641: tft35 e3 v3.0.1 + skr min e3 v2 no printer attached/marlin mode knob not working HALP!

krompiel, 2022-11-25:
good morning gentleman.

after exhaustive searches and videos and tears and wires and crimps and neglecting my kids... you are my last hope.

i recently purchased a skr mini e3 v2 and tft35 e3 v3.0.1 te replace burnt board in my anycubic i3 mega s.super silent board, super great. bl touch working just fine etc. problems arose with the screen though. i cant seem to get the bloody thing to connect in tft mode ( sais no printer attached) and in marlin mode, the home screen works and you can turn the knob to increase speed but the moment you push it it resets the printer.

actions taken: 
reflashed fresh firmware on screen
made new cable for tft
reversed cable for marlin mode
#define serial_port 2
#define serial port 2 -1
stock cr10 display in config
tried both baudrates and 250000 and 112500 screen and board

please find attached my configs (even though i am a months deep and multpile revisions in, this is still my first firmware expedition)
the tft35 config is unmodified version from bigtreetech_gd_tft35_e3_v3.0_v27.x mar 10 2022
[confeegs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10093635/confeegs.zip)

i have contacted btt support however the first reply (32 hours later) said he needed to ask a colleague, at which point i decided to take advantage and attached configs and pictures and video to the email. "justin" then thanked me and asked for my details which im not very comfortable with seeing that i already had to enter it to open the ticket (is this kosher?). aaaanyway that is where we are at now.

thank you in advance for your time. it is truly appreciated.

radek8, 2022-11-25:
Can you please attach pictures of the wiring?

krompiel, 2022-11-26:
with pleasure friend. these pics are of stock and then a replacement wire i tried. green is reset
[image]
[image]
[image]
[image]

thank you

radek8, 2022-11-26:
Your description of the problem does not match the attached configuration.
You write, you set the word:
#define serial_port 2
#define serial port 2 -1

But the attached configuration says otherwise (that's wrong).
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

There are different cables in the photos. Which picture is correct? Or do you have two printers?

You did not attach the configuration file for TFT config.ini

You must provide accurate and truthful information. Otherwise it's hard to help you.

radek8, 2022-11-26:
You write that pressing the encoder resets the printer.
Does the reset only take place on the SKR and the TFT does not reset and continues to display?
Or does the TFT also reset along with the SKR?

radek8, 2022-11-26:
Can you check the TFT with a magnifying glass?
Don't you have these 2 pins shorted?

[image]

krompiel, 2022-11-27:
evening mr radek.

i apologize for the earlier confusion. on the config file i changed it because it made no difference when i tried it. i changed it back again just for the sake of argument and after this post ill reflash it ( in the interest of replying as fast as possible to thank you for your effort on this fine saturday night).

as for the two different colored wires, it was merely to illustrate that there was a second cable made. it is the same screen and board

as for the encoder. please see attached video. it does not seem to restart the board but it does lose connection with th epc and after the reset it tests/pulses the hotend fan like when you turn it on.

i peeled the tape back above the screen to take a look at them pins. they are soldered pretty cleanly without any notion of shorting. ill attach a pic however its hard to get it focused enough to be able to tell definitively on a screen.

[configspics.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10097310/configspics.zip)

once again thank you.

radek8, 2022-11-27:
According to the video only Merlin resets on SKR.
Did you make any modifications to the pin file for thle board?
So on the files:
pins_BTT_SKR_MINI_E3_V2_0.h
or
pins_BTT_SKR_MINI_E3_common.h

radek8, 2022-11-27:
Also check the SKR board for a short between the pins.

[image]

krompiel, 2022-11-27:
good afternoon mr radek.

no short apparent on board. even though i can not say for certain, i doubt i modified any pins files. i scrolled through them quickly and it looks fine however the pin assinment for the rest pin in common.h looks weird.

i do appreciate your effort but if this doesnt fix it ill be happy enough as is. no need to bother with me further. at least it is functional now in tft mode.
[pins.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10099002/pins.zip)

radek8, 2022-12-31:
After switching to Marlin mode (by holding your finger on the screen, and measuring Marlin mode), touching the first upper corner has the same effect as clicking on the encoder.
Does it also reset your printer?

oldman4U, 2023-01-09:
What is the status of this ticket? Please help the community and close the ticket once you do not need it anymore. Thank you

krompiel, 2023-01-09:
Good morning sir.

I hope to try what radek said tonight, I have been a bit preoccupied and apologize for my non responsiveness.

After I try it I will update/close the ticket.

Thank you.

Nico van Staden
Plant Manager
1552 rue Nationale, Terrebonne, QC  J6W 6M1 | www.ataraequipment.com<http://www.ataraequipment.com/>
':1-866-931-5445 x105 | 7:514-931-0629 | •: ***@***.******@***.***>
[Description: Logo-Atara]

krompiel, 2023-01-10:
Good evening Mr Radek. I followed your instructions and it does indeed reset the screen and printer when I touch the upper right corner in marlin mode.

radek8, 2023-01-10:
So the problem is not directly in the encoder.
Please try the following.
Disconnect the EXP connector 10 pin cable.
Connect only the 5 pin connector of the TFT connector. All 5 pins must be connected to both TFT and SKR.
Turn on the printer. after booting the printer in touch mode, press the encoder.
Has there been a reboot?
If so, the problem is a short circuit or the CPU of the display.
If not, the problem is a short circuit or the CPU on the SKR board.

guruathwal, 2023-01-15:
There was a similar issue a few years back. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/964

krompiel, 2023-02-28:
> So the problem is not directly in the encoder. Please try the following. Disconnect the EXP connector 10 pin cable. Connect only the 5 pin connector of the TFT connector. All 5 pins must be [...]

it did indeed reboot.

thank you very much sir. you have been a great help.

krompiel, 2023-02-28:
thank you to everybody that helped. it is dearly appreciated.

closed 2023-02-28 by krompiel