bigtreetech/BIGTREETECH-TouchScreenFirmware #2644: [BUG] Incorrect interpretation of temperature data in dual extruder setup

rondlh, 2022-12-04:
In a setup with 2 (or more) hotends, Marlin reports temperature and target temp for: T, T0, T1 etc., where T is the temperature of the **active extruder**, but the TFT always interprets T as the T0 temperature. This results in incorrect temperature readings when T0 is no active.

oldman4U, 2022-12-05:
Duplicate of #2637?

kisslorand, 2022-12-05:
I found the issue, might have a PR for it later today.

oldman4U, 2022-12-05:
👍🏻

rondlh, 2022-12-05:
The cause is obviously in parseACK.c

```
    else if ((ack_seen("@") && ack_seen("T:")) || ack_seen("T0:"))
    {
      heatSetCurrentTemp(NOZZLE0, ack_value() + 0.5f);
      heatSetTargetTemp(NOZZLE0, ack_second_value() + 0.5f, FROM_HOST);

      for (uint8_t i = 1; i < MAX_HEATER_COUNT; i++)
      {
        if (!heaterDisplayIsValid(i))
          continue;

        if (ack_seen(heaterID[i]))
        {
          heatSetCurrentTemp(i, ack_value() + 0.5f);
          heatSetTargetTemp(i, ack_second_value() + 0.5f, FROM_HOST);
        }
      }
```
A simple solution would be to let the for loop start from 0.
`
      for (uint8_t i = 0; i < MAX_HEATER_COUNT; i++)
`

kisslorand, 2022-12-06:
Check PR #2647.

rondlh, 2022-12-06:
> Check PR #2647.

Just tested it on an IDEX system and a single extruder system. Both work fine now, great job!

rondlh, 2022-12-08:
Fixed

PR #2647 (2022-12-28): Fix temperature parse when multi extruder
### Requirements

BTT or MKS TFT

### Description

On multi extruder printers parsing of extruder temperature is broken.
This PR fixes it.

### Related Issues

Fixes #2644
Possibly fixes #2637 too.
closed 2022-12-08 by rondlh