bigtreetech/BIGTREETECH-TouchScreenFirmware #2645: [BUG] Setting the stepper current for extruders doesn't work (Marlin)

rondlh, 2022-12-04:
Using the settings menu to set the stepper current for an extruder (E0, E1) doesn't work.
I found that if I try to set the current for E0 the command send is incorrect (M906 T1 E123), this sets the current for E1.
If I try to set the current for E1 nothing happens (no command is send to Marlin).
Environment is Marlin 2.1.x (current latest).

I confirmed that setting the current for X, X2, Y, Y2, Z, Z1, Z2, Z3 and Z4  all work fine.

kisslorand, 2022-12-04:
Check PR #2646.

rondlh, 2022-12-05:
> Check PR #2646.

That fixes the issue, checked for both E0 and E1, well done!
It's surprising to see that the code compiles with the commas in the wrong place.

PR #2646 (2022-12-28): Fix parameter settings
### Requirements

BTT or MKS TFT

### Description

Setting from the TFT's machine settings menu the driver current or TMC hybrid threshold for the extruder(s) is broken. This PR fixes it.

### Related Issues

Fixes #2645