bigtreetech/BIGTREETECH-TouchScreenFirmware #2658: TFT70 screen is reversed from printer movements.

mortso, 2022-12-29:
Using a TFT70 v3 linked to a Bigtreetech Octopus pro 1.0 board via EXP1 and EXP2.  This is on a Zaribo ZLC printer which is a CoreXY with a frame mounted bed and the print gantry moving all directions including Z. 

The display is reversed from the printer movements, in that moving the left arrow on the X axis moves the printhead to the right, on the Y the up arrow moves the print head back to you instead of away, and the Z is probably the same on the left 2 motors but the right 2 motors move inversely, so trying to raise the Z gantry causes the right side to raise and the left to lower. The cables are checked and correctly inserted in the Octopus board and the motors. 

So 2 questions: How to reverse the display and the printer main board so they're headed the same directions and (2) is there an obvious way to get 2 of the 4 Z motors to reverse and match the other 2?

radek8, 2022-12-29:
For the wording of the direction of the X and Y axis, edit the item in the config .ini file
  inverted_axis:X1 Y1 Z0 LY0
or
inverted_axis:X1 Y1 Z0 LY1
or
inverted_axis:X1 Y1 Z0 L1

To unify the direction of rotation of the Z-axis motors, swap the two terminals on one motor drive. This will change the rotation direction of the motor without having to upload a new FW.

[image]

mortso, 2022-12-30:
Thank you, however I'm confused about the LY? protocol or wording; can you explain what the 3 choices affect? 
(Sorry, am not an experienced coder, just a hack still learning.)  So LY0 ; LY1 ; L1 indicate what? 

The printer I am working with has a stable unmoving print bed and the Y axis is on the CoreXY gantry supporting the print head, the entire gantry is on linear bearings and moves up and down via 4 motors and belts.  So the natural movement of Y would be arrow up; go back (+) arrow DOWN move forward (-) with all home positions being a X,0; Y,0; Z0 

Thank you in advance, I did the wiring changes and now all the motors are in sync. and moving the gantry same direction :-) Here is a pic of the printer for clarification. 

[image]

radek8, 2022-12-31:
L1 is nothing, it's my fault.
X0 and X1 change the direction of X-axis movement during manual X-axis movement.
Y0 and Y1 change the Y-axis movement direction during manual Y-axis movement.
LY0 and LY1 change the Y-axis movement direction in the manual bed leveling menu.

oldman4U, 2023-01-01:
Nice printer!

Please help the community and close the ticket once you do not need it anymore. Thank you

mortso, 2023-01-02:
THANK you for your help.... I may have a continued question so I'll leave it open for the next day or so. 

I changed the Invert settings in config.ini and now, in TFT mode it's going the  _right direction physically_  however it's positive and negative values are reversed. As well, in Marlin it's still moving the motors counter to the numerical plus / minus values; so that "home" seems to be the back right of the printer instead of the left front. So moving the printer back _TOWARDS_ actual home, the numerical indication goes to _positive_ numbers instead of back to zero. It's the same with X, Y and Z. 

I dare not do a home command not sure what would happen! Is there a firmware setting in the Octopus Pro 1.0 board perhaps that needs a jumper switched or such? Changing all the pins in the wires as per above is an option but is there another solution more obvious?

radek8, 2023-01-02:
Yes, there is. But that's where you should have started. 
You need to change the direction of rotation of the stepper motors in Marlin and perform a new Marlin compilation.

closed 2023-01-07 by mortso