bigtreetech/BIGTREETECH-TouchScreenFirmware #2659: [BUG] The "chamber" temperature is not updated in "RRF" mode.

langyaWang22, 2022-12-30:
The "chamber" temperature is not updated in "RRF" mode.The "chamber" will be updated only when the screen is turned on or restarted.
TFT Firmware Version: new master - BIQU_TFT35_V1.2.27.x.bin

radek8, 2022-12-31:
This will probably be a communication problem with RRF.
This is the query for @Subcircuit , @pfn , @bttguy,

KirkBigelow, 2023-01-28:
Have an Octopus Pro 1.0.1 and GD TFT50 V3.0.1.  Have similar issue with display not updating with RRF config.  Have run through all the possible serial mode with modes S0 S2 showing the printer is connected at baud rate of 57600.  RRF firmware 3.4.5_102 and TFT firmware V3.0.27.x.  It seems to communicate only after a reset and then gets lost.  The TFT terminal will not do anything either.   The Octopus Pro seems to be work fine with the limited test done so far.  Octopus Wifi interface and Duet Web page dashboard is as expected as well as the Octupus command USB interface.

closed 2023-04-25 by stale[bot]