bigtreetech/BIGTREETECH-TouchScreenFirmware #2663: Flashing TFT35 E3 V3 (Gigadevice MCU) with St-Link

n30V13l, 2023-01-02:
Hi!

I kind of bricked my TFT35 E3 V3.01 (with GD MCU) by using the STM32 and NOT the GigaDevice firmware .bin and flashing via SD-CARD.
It was then stuck with the "illegal APP" Message even after mutliple reboot and did all the troubleshooting measures mentioned in the offical guide, with no success.
In order to "unbrick" I built the new firmware with platform.io ( BIGTREE_GD_TFT35_E3_V3_0 )and flashed the binary via SWD-Interface and ST-Link v2/ST-Programmer.
But now the TFT Screen keeps black, nothing happens. 
Do I have to use the GD-Link instead of ST-Link? 
How can I "unbrick" my TFT35? # #

radek8, 2023-01-03:
If nothing helps, you can try this
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/discussions/2649

n30V13l, 2023-01-06:
Thanks a lot! This helped.

oldman4U, 2023-01-08:
Please help the community and close the ticket once you do not need it anymore. Thank you

n30V13l, 2023-01-08:
> Please help the community and close the ticket once you do not need it anymore. Thank you 

Sorry, thought I did. Must have been a brain malfunction ;)

closed 2023-01-08 by n30V13l

oldman4U, 2023-01-08:
👍🏻😁