bigtreetech/BIGTREETECH-TouchScreenFirmware #2666: [Q] Is it possible to show extruder position during print

alxsfrm, 2023-01-07:
Hello,
I build the firmware by myself.
I have latest Marlin on SKR mini E3 + TFT35 V3 E3.
I have following issue: before printing i can see in the middle of the screen the status line with extruder XYZ values, but when i'm starting to print from the onboard SD, the status line disappear, and instead of it there is progress bar.
Also the layer number/height is not shown, just ------ during all the printing process.
How can i make the extruder position and the layer height to be visible during printing ?
Thank you.

kisslorand, 2023-01-07:
For the layer number height issue:  - do you have the necessary options enabled in Marlin?
For the extrusion position during print issue: - there's no such option implemented in the current TFT FW, you would need to modify the source code to fulfill your needs

alxsfrm, 2023-01-07:
> For the layer number height issue: - do you have the necessary options enabled in Marlin?

May be i missed something. Can you please point me to the specific option needed to enable this ?

kisslorand, 2023-01-07:
Check if you have enabled AUTO_REPORT_POSITION (in Configuration_adv.h)

radek8, 2023-01-09:
Why do we need to know the position of the extruder when printing when it is constantly changing during printing?
Isn't that useless information?

alxsfrm, 2023-01-13:
Actually you are right. I just used to look on it in order to see the layer height.

alxsfrm, 2023-01-13:
> Check if you have enabled AUTO_REPORT_POSITION (in Configuration_adv.h)

Yes, it is defined. Still no layer number/height.

kisslorand, 2023-01-23:
Also check if you have enabled EXTENDED_CAPABILITIES_REPORT  (in Configuration_adv.h)

LTSpenny, 2023-02-03:
@alxsfrm i solved this by adding a gcode report after layer change. 

Mine is 
;BEFORE_LAYER_CHANGE
;LAYER:[layer_num]

Hope that helps.

oldman4U, 2023-02-10:
What is the status of this ticket?

closed 2023-05-03 by stale[bot]