bigtreetech/BIGTREETECH-TouchScreenFirmware #2670: [BUG] TFT24 v1.1 BTT-TouchScreenFirmware-master20230113 Fails to Process Config.ini File

helakejr, 2023-01-14:
TFT24 v1.1 BTT-TouchScreenFirmware-master20230113 Fails to Process Config.ini File

Use of the TFT24 v1.1 firmware, config.ini, and TFT24 directory from https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

Steps to reproduce:
Need: 
- TFT24, SD Card containing TFT24 v1.1 firmware BIGTREE_TFT24_V1.1.27.x.bin, config.ini, and TFT24 directory
- BTT SKR Mini E3, SD Card containing firmware from my Anet AM8 Marlin build (flash Marlin build)
- BTT SKR Mini E3 V3.0, TFT24 V1.1, 5-pin serial cable (touch-mode), 10-Pin to two EXP1/EXP2 cable from BTT.

1) Insert TFT24 SD Card
2) Apply power
3) Results: Loading Fonts, Logo, Icons
- Error messages finding U_Disk.bmp, Marlin.bmp, BIGTREETECH.bmp
4) loading config.ini
- Errors "Invalid Keyword:" pop-up errors

Refer to attached video documenting the results. (Too large for this forum)
https://drive.google.com/file/d/1K5Qugl_I5cy0wLR9mvMIJ9R1TZbbzNlz/view?usp=share_link

guruathwal, 2023-01-15:
Your firmware did not get updated. re-check if the firmware file gets copied on the sd card. also, note that there are two variants of  TFT24 one with an STM32 chip and the other with a GD chip. Make sure you have the correct one copied.

helakejr, 2023-01-15:
The firmware file was not renamed.CUR. The config.ini and TFT24 were
renamed .CUR.

helakejr, 2023-01-15:
Discovered I have the GD arm chip.
Updated sd card to use GD version of firmware.
All updates: firmware, fonts, icons and config.ini now error free.

Thank you for letting me know.

guruathwal, 2023-01-16:
Good. you can close this issue now.

closed 2023-01-16 by helakejr