bigtreetech/BIGTREETECH-TouchScreenFirmware #2681: [BUG] TFT70 + SKR PRO 1.2 (no printer attached)

syszone, 2023-01-20:
I am upgrading my printer with SKR Pro 1.2 with TFT70. Earlier it was working then i have upgraded the configuration as per TouchScreen suggestion on : "General options which MUST be always activated" since then i see "No printer attached"  even i have reverted back the the changes with no luck. 

I am attaching configuration.h, configuration_adv.h and config.ini. Hopefully someone can look at the file and suggest the solution.

Thanks

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10466241/Configuration.zip)
[image]
[image]

digant73, 2023-01-20:
check and eventually set the TFT baudrate (Menu->Settings->Connection->S. Ports menu) so it matches the baudrate you configured in Marlin.

syszone, 2023-01-20:
in Marlin my serial port is 1 
#define SERIAL_PORT 1
TFT baud rate is set to 250000

radek8, 2023-01-20:
> in Marlin my serial port is 1 #define SERIAL_PORT 1 TFT baud rate is set to 250000

Yes, Marlin is set to 250000, but on TFT you set 115000 in the .ini file

syszone, 2023-01-21:
I have changed to port 7 in config.ini, still same
serial_port:P1:7 P2:0 P3:0 P4:0 
[image]

radek8, 2023-01-21:
Check if you have the serial cable between SKR and TFT plugged into the correct connector and if it is oriented correctly.

syszone, 2023-01-21:
serial cable connected correctly, it was working before I update the marlin with suggested changes by TFT main page, even i try to revert them I do not know what I am missing.

kisslorand, 2023-01-22:
Try with the following settings in "Configuration.h":
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
```

syszone, 2023-01-22:
@kisslorand it did not work

kisslorand, 2023-01-23:
Is there any chance you flipped the TFT cable? Do you have another TFT to test with? Can you access the mainboard via USB (Octoprint, Printerface, etc)?

syszone, 2023-01-30:
Flipping cable wont start TFT it will be blank screen, and Marlin mode shows only blank screen. I am trying to connect USB but on my computer it shows Unknown USB Device (Device descriptor request failed) i am using windows 11 if that helps.

kisslorand, 2023-01-30:
You might have a broken mainboard...

oldman4U, 2023-02-02:
Any news regarding the reported issue?

Please help the community and close the ticket once you do not need it anymore. Thank you

syszone, 2023-02-03:
@kisslorand i guess you are correct. I will investigate my own on fuse maybe or order new board. Closing this thread. Thank you everyone for try to helping out - cheers

closed 2023-02-03 by syszone