bigtreetech/BIGTREETECH-TouchScreenFirmware #2684: I cannot level my printer in marlin mode  without bltouch!

DavidFFerreira, 2023-01-22:
I cannot level my printer in marlin mode  without bltouch!

radek8, 2023-01-27:
How does working in Marlin mode relate to FW TFT display?

oldman4U, 2023-01-28:
Please help the community and close the ticket once you do not need it anymore. Thank you

DavidFFerreira, 2023-01-28:
marlin configuration is in your board firmware...and that firmware have bltouch enabled....since you have it enabled, i cant level my bed on marlin because it require bltouch

radek8, 2023-01-29:
Yes, if you have BL-Touch enabled in the Marlin, but it is not physically connected, you cannot level the bed.
So you have to compile a new Marlin.
I ask again, how does this relate to the FW TFT display?
You can deactivate the automatic bed leveling in the display configuration, but this will not solve your problem in Marlin.

oldman4U, 2023-01-29:
David. Radek8 asks how this is related to the FW because this is the BTT TFT firmware repository but not a user group.

oldman4U, 2023-02-02:
@Msq001 Could you please check and close.  Thank you

closed 2023-02-10 by bigtreetech