bigtreetech/BIGTREETECH-TouchScreenFirmware #2688: I cant load the config file[BUG] (short description)

changwang31, 2023-01-25:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2023-01-27:
Can you provide more information on your problem?

oldman4U, 2023-02-02:
@Msq001 could you please check and close

closed 2023-02-09 by bigtreetech

changwang31, 2023-02-11:
I fixed it 
in the readme it said to load the file as config.ini
but my model (tft35 e3 V3.0)
wouldn't read it
so i just tried something new and loaded it without the suffix and it worked

changwang31, 2023-02-11:
and sorry for not closing it 
im new to this community