bigtreetech/BIGTREETECH-TouchScreenFirmware #2691: sensor not recognized by Klipper

MartinM8266, 2023-01-26:
Hello, is there a way for Klipper with BKS Mini E3 V3.0 and TFT35 V3.0 ?
Klipper console isnt´t recognizing filament motion sensor when connected to PC15 (E0 endstop connector). thanks!

radek8, 2023-01-27:
That sounds logical.
Klipper cannot see the PC15 pin.
PC15 pin can detect FW TFT if printing from TFT.
But if you are printing from Klipper, the PC15 pin is unavailable to Klipper.

MartinM8266, 2023-01-27:
aha. but Klipper indeed can see the other pins on the BKS Mini (e.g. stepper pins, endstop pins x,y,z). Why not the PC15 (endstop E0)?
I am not printing through the TFT35 (sd-card). I am printing via Mainsail running on Raspi3. 
Raspi 3 connection to BKS Mini via USB (not UART). 
the TFT35 is only connected to the BKS Mini via serial connection only (no touchscreen-cable) and runs in emulated 12864 (stock display) mode.
I do not use the filament-detection-port on the TFT35 ( I understood this is not working with Klipper, but have tried though - it is not working).
thanks for help.

radek8, 2023-01-27:
Are we talking about the same PC15 pin?
I am talking about pin PC15 which is on TFT35.

MartinM8266, 2023-01-27:
No sorry - I am talking about the PC15 on the BKS Mini E3 V3 (is is the E0-endstop).
For the pins on the TFT35 it is clear that Klipper does not "see" them.

radek8, 2023-01-27:
So why are you asking here? This is the FW storage for TFT.

MartinM8266, 2023-01-28:
I thought it might be possible to map thenpin through somehow. thanks anyway.

closed 2023-01-28 by MartinM8266