bigtreetech/BIGTREETECH-TouchScreenFirmware #2697: SKR 2 & TFT70 V3.0 - No Printer Attached!

Unhallowed1949, 2023-02-03:
I know that "No Printer Attached!" has been covered a million times but I've spent at least 20 hours, probably a lot more, reading and learning and editing and I can't get the SKR 2 and the TFT70 to talk.

### What have I done and verified?

EXP1 goes to EXP1 and EXP2 goes to EXP2 
The RS232 cable is absolutely plugged in the correct way. I've checked this many times and used my multimeter to check continuity to make sure the cables are good. RST goes to RST, and the rest fall into place. I can confirm this 110%, no ifs ands or buts.

**Configured Marlin and TFT70 with the following combinations of ports and baudrates**

Marlin: "SERIAL_PORT -1" & "Serial_PORT2 1", and tried the baudrates defined together and separately at 250000.
TFT70 config.ini: serial_port:P1:7 P2:0 P3:0 P4:0 (7=250000)

Marlin: "SERIAL_PORT 1" & "Serial_PORT2 -1", and tried the baudrates defined together and separately at 250000.
TFT70 config.ini: serial_port:P1:7 P2:0 P3:0 P4:0 (7=250000)

Marlin: "SERIAL_PORT 1" & "Serial_PORT2 -1", and tried the baudrates defined together and separately at 115200.
TFT70 config.ini: serial_port:P1:6 P2:0 P3:0 P4:0 (6=115200)

Marlin: "SERIAL_PORT -1" & "Serial_PORT2 1", and tried the baudrates defined together and separately at 115200.
TFT70 config.ini: serial_port:P1:6 P2:0 P3:0 P4:0 (6=115200)

I tried setting "SERIAL_PORT 0" just for the hell of it and it wouldn't compile.

I even tried mismatching 115200 and 250000 because some random guy said it worked for him. I assumed this would not work and it didn't.

In Marlin, I have defined  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and no other displays are defined. I've also defined every other setting listed on the github, there are a total of 7 required definitions and 17 optional definitions and I have enabled them all, except for Z_STEPPER_AUTO_ALIGN because currently, I'm not using two Z steppers.

**Updated the TFT 70 Firmware using:**
BIGTREE_GD_TFT70_V3.0.27.x.bin (my chip is GD)
TFT70 Folder (for the circle theme)
config.ini (with baudrate set to 115200 or 25000, depending on how I set up Marlin for that attempt)
language_en.ini
I had an error during flashing "invalid keyword - label_shim:shim", followed by a screen that said "error: invalid config file". I believe this is only related to the language portion of the flash because all files and folders changed to .CUR except for the language file. After that, I opened the language_en.ini and commented out "label_shim:shim" and tried to flash the language_en.ini file again. Same error, but no invalid keywords. 

I tried sending reset.txt to the TFT to hard reset it (successfully), and then redid all of the above, no luck.

The github states that the  "No Printer Attached!" message is reliant on 3 things; connect serial cable correctly, define marlin ports correctly, and have matching baudrates. I have done all 3 of these things correctly!

I tried putting 1k resistors in place of the thermistors because someone said the board might be in a fatal error due to no thermistors and causing issues. I measured out my thermistors at room temp and they were roughly 1k, that's how I chose this value. This did not work.

Lastly, if I switch to Marlin Mode, the screen is blank except for a message at the top that says "Marlin Mode"

Can someone please help me figure out what is going on? My understanding is that it should be working the way I have it configured.

radek8, 2023-02-03:
Hi.
Yes, as you describe it should work.
I noticed a few things in your description.
You write that the printer does not connect in touch mode, but you also write that the printer does not display anything in Marlin mode.
Do I understand correctly that your display does not work in either mode?

- If you upload a new FW Marlin to SKR, will the bin file be renamed to cur?
- Do you have another display with which you can test the functional mode of Marlin?
- Does the printer respond to commands from the computer if you connect it via USB?

Unhallowed1949, 2023-02-03:
To answer your questions, yes they change to .cur, I tried a TFT35 and TFT70, I have no tried via USB but I have fixed the issue and I think the github section that tells us why the message "No Printer Attached!" would pop up should be updated to include the following scenario.

I had the following definitions
temp_sensor_0 1047
temp_sensor_bed 1
temp_sensor_chamber 1047

changing those definitions to the following
temp_sensor_0 1047
temp_sensor_bed 1
temp_sensor_chamber 0
and plugging in an extra PT1000 I had laying around into the hotend port caused the "No Printer Attached!" message to go away!

I have done a few more tests and have concluded that any sensor defined as 1047 without a PT1000 plugged into its defined port will cause the "No Printer Attached!" message to be displayed. 

Having any of the temp sensors defined as 1 and not having a thermistor plugged in will not cause the "No Printer Attached!" to be displayed.

I'm not sure if defining temp sensors as other sensors will also cause the message but I think it would be great if the github mentioned that not having thermistors plugged into the ports can cause the "No Printer Attached!" message to be displayed. I've been chasing my tail for days and that simple statement would have saved me a lot of time.

Let me know what you think and then I will close this ticket.

radek8, 2023-02-03:
From what you write, the problem was not with the TFT display, but with the Marlin configuration.
Due to a bad configuration, Marlin did not work for you and therefore could not communicate with the display.
A working Marlin is required for the connected display.
The question is why Marlin does not work if it has a sensor 1047 defined and this sensor is not connected.
Maybe you could test this issue more and possibly report the issue on Marlin's GutHub.

Unhallowed1949, 2023-02-04:
Yes, that is exactly what I found out. I will report the issue to Marlin. This only happens with temp_sensor defined as 1047, defined as 1, the thermistor does not need to be plugged in.

I was just suggesting that the section in the BTT github about "No Printer Attached!" should be updated to include that if any of the temp_sensors are defined as 1047 and the thermistor is not plugged in, this will also cause the message "No Printer Attached!" to be displayed, not just the 3 variables mentioned in this github. If any of the temp_sensors are defined as 1, the thermistor can be unplugged and the message "No Printer Attached!" **_will not_** be displayed. This leads me to believe there is a bug in Marlin. 

I'm just hoping to save people a lot of time chasing their tail, like I did, and I think it's worth mentioning.

Cheers!

closed 2023-02-04 by Unhallowed1949