bigtreetech/BIGTREETECH-TouchScreenFirmware #2698: [BUG] Filament run out sensor not working with tft mode

labotecno, 2023-02-04:


LTSpenny, 2023-02-06:
i don't think this is enough information. The tft has it's own input for the filament run out sensor that you can plug directly into it and set up. When you enable filament runout sensor on the tft settings it's referencing that sensor and setup. If you have your filament runout sensor set up to the main board, then those settings activate regardless of what your tft display mode is in. Have you ensured the correct settings are enabled in marlin? also information useful for this is expected behavior and what you are experiencing. hard to assume what is happening with given information.

labotecno, 2023-02-06:
> 
my mainboard is skr e3 turbo .
i have no option runout filament sensor  enable from tft . runout filament sensor is connected to the mainboard , my config is Working with marlin mode 
i try to put the gcod M75 and M77 start /end gcod on the slicer and on the tft option too . may be mean serial don't recognize it . i know i can to connecting sensor directly to the tft , but i want to make it working from mainboard . thank you to help me

LTSpenny, 2023-02-06:
What release are you using. What tft do you have. Im running a tft35 and 43 and i do not use those gcodes in my start and the filament runout sensor works fine. If my filament runs out it triggers an advance pause/filament change. Please tell is what is happening. What happens. What you expect to happen and what it is doing that is not expected.

labotecno, 2023-02-06:
> What release are you using. What tft do you have. Im running a tft35 and 43 and i do not use those gcodes in my start and the

its make nothing , when the filament is cut it dont stop or make pause just printing  . i use the last release of the firmware from here i have tft35 E3 . do you have your runout filament connected to the  mainboard ? and i have the last marlin firmware bugfix

LTSpenny, 2023-02-06:
Can you post your config file for your marlin bugfix. How are you printing files? Inserting them into tft or mainboard?

labotecno, 2023-02-07:
configuration.h 
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10670741/Configuration_adv.txt)
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10670742/Configuration.txt)
i print from main board sd card , and i try  too from tft lcd too

labotecno, 2023-02-11:
someone have a idea ?

LTSpenny, 2023-02-12:
I looked through your code and it all seem’s good. I highly doubt it is an issue with the tft as the tft just displays what the main board is providing unless you have a runout sensor plugged directly into it. I also do not include those gcodes in my sliced files and dont have issues on my two machines with the tfts. Are you using the master files or older release?

purplewazard, 2023-02-16:
im also having the same problem i cant get it to work in tft mode when connected to the board or the tft

oldman4U, 2023-02-25:
Labotecno. Could you please take care of your ticket and answer the replies. Thank you

labotecno, 2023-02-25:
> I looked through your code and it all seem’s good. I highly doubt it is an issue with the tft as the tft just displays what the main board is providing unless you have a runout sensor plugged [...]

I m with bugfix last version .

closed 2023-05-08 by stale[bot]