bigtreetech/BIGTREETECH-TouchScreenFirmware #2703: How can i disable the "some heaters are still on" Message?

Primus007, 2023-02-07:
My goal is to disable the message "some heaters are still on". This message popup on my display when i have preheated the nozzle/bed and go to the Menue Movement -> Bed Level and then Back.

I have a new TFT35 E3 V3 Display. 
The Source is the Vx.x.27 from here. 
I have disabled the SAFETY_ALERT in the configuration.h and compiled and updated the Firmware on the display.
but the  message still popup. 

How can i disable this message? Why it does not work if i disable SAFETY_ALERT?

radek8, 2023-02-07:
The setting in configuration.h is the default, but parameters in config.ini override it.
Did you use Config.ini?

Primus007, 2023-02-07:
after the reset of the display i have not used a config.ini
And the SAFETY_ALERT is not in the config.ini. This can i edit only in the configuration.h
the display always shows "BIGTREE_GD_TFT35_E3_V3_0 Firmware V27.x Mar 10 2022 in 120MHz" in the info. shouldn't be the current date if the update was successful?

radek8, 2023-02-07:
Use the current branch. Master
Everything else is obsolete

[image]

radek8, 2023-02-07:
Coment A SAFETY_ALERT 

[BIGTREE_GD_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/10679795/BIGTREE_GD_TFT35_V3.0_E3.27.x.zip)

radek8, 2023-02-07:
When updating, you can see the progress of the update on the display.
After the update, the file extension will change from bin to cur
Also update the ini file, language file, and icon and font folder.

Primus007, 2023-02-08:
i have tried your bin. it does not update. My display does not update any .bin

Now i have noticed that my display does not update the Firmware at all.
i copy the firmware to an sd card.
then i put the sd card in the display and boot with it.
I don't see an update screen
But the firmware was renamed to .cur
I have tried different SD cards
i have tried precompiled firmware and self compiled. 
all other things will be updated. config.ini, themes and language files. Only the .bin not.

i use this bin: BIGTREE_GD_TFT35_V3.0_E3.27.x.bin

here is a photo from the info screen:
[image]

radek8, 2023-02-08:
Is your display really the GD version?
Can you check the CPU model on the display?
Send a photo of the circuit board.

radek8, 2023-02-08:
SD card should be 8G or smaller.
Formatted as FAT32 (small capacity cards as FAT)

Primus007, 2023-02-08:
The display is new. it is the original firmware you see on the picture above.
I check the CUP later and upload a photo.
I have tried different sd cards. 
it is strange that the file is renamed but the firmware is not updated.

Primus007, 2023-02-08:
Yes it is a GD Display. I have tried different sd cards. it is strange that the file is renamed but the firmware is not updated.
[image]

[image]

radek8, 2023-02-08:
Yes, you have the correct version (GD)
So far no one else has complained about the same problem...

radek8, 2023-02-08:
Yes, you have the correct version (GD) I would try to ask for a statement from the manufacturer. On his website, he occasionally responds to user inquiries.
The manufacturer does not respond to questions on Github.

https://support.biqu3d.com/hc/en-us/sections/4415370318745-TFT35-E3-V3-0

Primus007, 2023-02-08:
Thanks for your help!

radek8, 2023-02-08:
https://community.biqu3d.com/forum/40-product-discussion/