bigtreetech/BIGTREETECH-TouchScreenFirmware #2716: "invalid Keyword" in Config.ini

Electromente, 2023-02-14:
Hi Guys, 

Im trying to configure my TFT35 V3.0.27.x with the stock firmware but I've modified the config.ini some times and everything has work fine. I lost the old config.ini file then I've downloaded the file from this repository.

The issue is when I try to apply this config file, the screen says a lot of "invalid keyword" warnings for several parameters. I.e,: serial_port:P1:6 P2:0 P3:0 P4:0, emulated_m600:1 and a lot of other fields.

[image]

The firmware repository (to copy to SD card) have a BIGTREE_TFT35_V3.0.27.x.bin uploaded 18 hours ago, but I don't know if this is a new firmware or is the same of the stock firmware, because I assume that a new firmware should say TFT35 V3.0.**28**.x or something like that. Also, the current installed firmware is working fine. I'm just trying to config the number of fans, take off some menus to be more efficient UI, etc.

Could you help me to know how to correct the config.ini file to take the right parameters? may be "emulated_m600:1" need some spaces emulated_m600: 1, or emulated word is with capital letter, like Emulated_m600: 1, I don't know.

By another hand, If I only apply the line that I want (i.e, fan_count:1) then I lose "custom" menu and appear in blank. Then I assume that I need to apply a complete config.ini file.

[image]

Please help!!! I just want to take off BLtouch, number of fans, change some colors, etc.

kisslorand, 2023-02-14:
Your best bet would be to use the latest firmware from the 'master' branch and its associated "config.ini". Don't forget to update the fonts, language and icon files too.

Electromente, 2023-02-14:
Thanks, I'll updated and and noticed that have a lot of changes also takes the config.ini.

closed 2023-02-14 by Electromente