bigtreetech/BIGTREETECH-TouchScreenFirmware #2719: [Q] MKS TFT32 V1.0 install firmware

stef-ladefense, 2023-02-16:
Hi,
I'm trying to install the latest firmware on a MKS TFT32 V1.0.
I followed the doc, renamed the MKS_TFT35_V1_0.27.x.bin file to mkstft35.bin, copied the TFT35 folder and the config.ini and copied everything to a 4GB SD.
the sequence begins then a progress bar reaches 100% and nothing more happens.
do you have an idea ?
Thanks a lot

digant73, 2023-02-16:
if you have a MKS TFT32 v1.4 you must use `MKS_TFT32_V1.4.27.x.bin` renamed to `MKSTFT32.bin` and the `TFT28` folder renamed to `TFT32` in order it can be recognized and installed by the TFT.

stef-ladefense, 2023-02-16:
Sorry, I made a typo
title : [Q] MKS TFT3**5** V1.0 install firmware
I'm trying to install the latest firmware on a MKS TFT3**5** V1.0.

it does not change the result, progress bar to 100% and no more reaction

stef-ladefense, 2023-02-17:
I copied the following files to the card
config.ini
MKS_TFT35_V1_0.27.x.bin renames MKSTFT35.bin
TFT35

and here are the progress pictures.

[image]

it hangs at 100% and after a reset, it does not boot

[image]

I place the photos of the MKS card

[image]
[image]

rondlh, 2023-02-19:
The MKS TFT35 V1.0 will erase all settings when you flash a new firmware, this might be the cause, perhaps it's in Marlin mode?
You could try to update the settings with a correct config.ini
Does the TFT rename your config.ini?

stuclem47, 2023-02-19:
Following this with interest.
I have exactly the same issue.
The file config.ini remains on the SD Card unchanged, as does the TFT35 folder.
Firmware file is renamed to: MKSTFT35.CUR

stuclem47, 2023-02-20:
I am coming to the conclusion that the firmware MKS_TFT35_V1_0.27.x.bin is not suitable for this model. I have tried flashing alternate bootloaders that have been suggested as workinfg elsewhere, but after this the firmware flash still fails.
There are several different models of MKS TFT35 screens, and this may be for one of the other models.
Has anyone actually got the BTT firmware loaded on this particular model?

stuclem47, 2023-02-20:
The MKS_TFT35_V1_0.27.x.bin (renamed as MKSTFT35.bin) flashes and shows progress to 100% and goes no further. File is renamed as MKSTFT35.CUR. It seems to looking for the mks_font & mks_pic folders and the mks_config.txt file, which are not part of the BTT package.

stuclem47, 2023-02-20:
Update:
Flashed open source bootloader from here: https://github.com/darkspr1te/mkstft35_V1_open_bootloader
Reamed the BTT file to mkslcd.bin as instructed there and added config.ini and TFT35 folders. The flash completed showing the icons on screen as they were uploaded, just like when my BTT TFT is updated. Then went to touch screen calibration, once this was completed it loaded a screen with just  one line of garbled text. Getting closer!

rondlh, 2023-02-21:
I just tried the conversion of a Original MKS TFT35 V1.0 (FW 1.0.7) to the 2023 Feb 16 BTT firmware.
All went as expected. Here the contents of my SD Card.
MKSTFT35.bin is the renamed MKS_TFT35_V1_0.27.x (CRC 01490B3F, 2023 Feb 16).
The TFT35 holds the files in "\Copy to SD Card root directory to update\THEME_Unified Menu Material theme\TFT35", both bmp and font directories.
[image]

Flashing worked as expected, all files and the dir get the extension ".CUR" added.

[image]

[image]

You could try another SD card, and make sure it's not write protected, that could cause issues.
darkspri1te is the creator of the MKS TFT35 V1.0 port of the BTT firmware, which is greatly appreciated!

I'm not sure if I'm allowed to post the original MKS TFT35 V1.0 bootloader here, I have both the original version and a version I patched so you don't need to "encrypt" the firmware anymore. If you use the wrong version the application will not work (black screen), but it has no effect on the flashing process.

stef-ladefense, 2023-02-21:
> I have the MKS TFT35 V1.0, from the picture I would say it's exactly the same. I can give it a try. One of the pictures shows "Firmware: 1.0.8", that's the original MKS firmware, not the BTT hosted here. It will NOT handle the BTT config.ini and BTT Icons.

hi,
it's pics of card BEFORE flash with this BTT firmware

stef-ladefense, 2023-02-21:
Hi,
I took the files like you, and same problem, 100% blocking without follow-up.
I then compiled the firmware (I had everything after my fix for the TFT32LV3.0) and I generated a 'MKS_TFT35_V1_0.27.x_encrypted.bin' file.
once renamed to 'MKSTFT35.bin' and copied to the SD, the update to the BTT firmware went through correctly.

[image]

https://github.com/stef-ladefense/BIGTREETECH-TouchScreenFirmware/commit/34195b95e86b7e9a67743f4edec6553ff0d2ef69

rondlh, 2023-02-21:
@stef-ladefense That's great news! Thanks for reporting back.

stuclem47, 2023-02-21:
@stef-ladefense 
Can you make a copy of the MKS_TFT35_V1_0.27.x_encrypted.bin file available?

stef-ladefense, 2023-02-21:
> @stef-ladefense Can you make a copy of the MKS_TFT35_V1_0.27.x_encrypted.bin file available?

allready in my github for moment
https://github.com/stef-ladefense/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

stuclem47, 2023-02-21:
Thank you @stef-ladefense 
I have now flashed the firmware successfully.

digant73, 2023-02-21:
the binary MKS_TFT35_V1_0.27.x.bin provided in this repo is the unencrypted version although I provided also the encrypted one. Currently, for the encrypted version you have to compile it by yourself as you did

oldman4U, 2023-02-21:
Please update the Readme if you would like to keep this information available for other users. Thank you

rondlh, 2023-02-23:
> the binary MKS_TFT35_V1_0.27.x.bin provided in this repo is the unencrypted version although I provided also the encrypted one. Currently, for the encrypted version you have to compile it by yourself as you did. Please close the ticket if you solved your issue.

I recommend to provide the encrypted FW by default, as that is what most users need.

digant73, 2023-02-24:
@rondlh fixed in #2728

PR #2728 (2023-04-11): Minor improvement on print abort
**IMPROVEMENTS:**
* Some code reduction: removed unused/not needed `storeEmergencyCmd()` and `handleEmergencyCmd()` on `interfaceCmd` API
* `handleCmd()` and `sendEmergencyCmd()` moved just before `sendQueueCmd()` on `interfaceCmd` API
* fast print abort applied also when printing from TFT or remote host: a break of any pending gcode (waiting for an ACK message) is also applied when printing from TFT and remote host
* `isPrintingFromHost()` renamed to `isPrintingFromOnboard()` (matching the naming convention used for a print from Marlin's onboard media

**BUGFI [... cut]
closed 2023-04-11 by bigtreetech