bigtreetech/BIGTREETECH-TouchScreenFirmware #2720: [BUG] MKS TFT32 L V3.0 Screen H Flip

stef-ladefense, 2023-02-17:
hi,
I just did a test on an MKS 32L V3.0 that I had in a drawer.

I simply took the following basic files:
config.ini
MKS_TFT28_V3.0.27.x.bin renames MKSTFT28.bin
folder TFT28

the MKS TFT32 shares the firmware of the MKS TFT28 and must therefore be named the same for the firmware update to work.

the update works until the end but the displays are mirrored horizontally and vertically at 180°.
at the end the touch adjustment procedure does not follow the inversions of the screen, it is therefore necessary to touch in the opposite diagonal.

[image]

I put the additional photos in the next post

stef-ladefense, 2023-02-17:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I hope it will help to fix the firmware

Stéphane

radek8, 2023-02-17:
Did you download the update files from the Marter branch?

radek8, 2023-02-17:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2436

stef-ladefense, 2023-02-17:
@radek8 
it's MKS display and the .bin has 2 day, 2.15.2023 10;53 from master

radek8, 2023-02-17:
Also try resetting the display with an empty reset.txt file

stef-ladefense, 2023-02-18:
I applied without effect

radek8, 2023-02-18:
The problem will be that your display is not among the supported versions of MKS displays

[image]

stef-ladefense, 2023-02-18:
yes but the MKS TFT32 V3.0 has the same hardware as the MKS TFT28 V3.0 so it shouldn't be very complicated to correct.
I will look at the sources and make the correction as soon as I have some time.

radek8, 2023-02-18:
MKS TFT 35 V1 uses
#define TFTLCD_DRIVER (ILI9488 | NT35310)

MKS TFT 28 V3 is used
#define TFTLCD_DRIVER ILI9325

radek8, 2023-02-18:
You can try it
MKS_TFT32_V1_3