bigtreetech/BIGTREETECH-TouchScreenFirmware #2724: [BUG] Stop or Pause print buttons not working for remote print

raihei, 2023-02-19:
### Description

When using remote print via Octoprint, the "Pause" or "Stop" buttons on the display do not send the notifications //action:notification remote pause or //action:notification remote cancel. Instead only a message "Busy processing, please wait" is shown on the top of the screen.

### Steps to reproduce

1. Start remote print or send M118 A1 P0 action:print_start to the display
2. Display switches to print screen
3. Press Pause or Stop button

**Expected behavior**
Display should send command  //action:notification remote cancel or //action:notification remote pause as described in the readme.md file.

**Actual behavior**
Commands are not being sent, only message "Busy processing, please wait" appears.

### Hardware Variant
BIGTREE_GD_TFT35_E3_V3_0
### TFT Firmware Version & Main Board Firmware details
current version compiled from github (27.x)
Mainboard: Marlin 2.1.2

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

raihei, 2023-02-19:
I checked the source code and to me it looks like, this feature is not implemented at all, but the readme states otherwise.

digant73, 2023-02-20:
if you installed the last TFT fw you should see a toast message displayed on notification bar when you press on pause/resume/stop buttons. Otherwise you are not using the last fw

raihei, 2023-02-20:
You are completely right, my apologies. I downloaded the source a few days before your change and did not notice that it has changed in the meantime. Also, I did not notice that this function is a new function which was recently introduced. I compiled the newest version and it works as intended. Sorry, again and thanks for implementing this function!

A quick question: Do you use OctoPrint and if yes, how did you set it to react to the "notification remote" commands? Or is this something, the printer (Marlin) should react to?

closed 2023-02-20 by raihei

digant73, 2023-02-21:
I don't use OctoPrint for printing (just for testing etc.). You have to configure on OctoPrint the remote notifications and react accordingly (e.g. pause a print). The TFT can only forward those notifications asking OctoPrint to perform an action.