bigtreetech/BIGTREETECH-TouchScreenFirmware #2727: [Q] raspberry pi os not work with touch pitft50 v1.0

Andrysky, 2023-02-21:
hi please tell me i have bigtreetech pitft50 v1.0
I installed raspberry pi os and connected pitft50 - the display worked out of the box, but the touch sensor does not work / bugs.
I installed the build from https://github.com/bigtreetech/OctoBTT - the sensor works fine.
What do I need to do on the stock firmware to make the touch sensor work?

oldman4U, 2023-02-21:
Hi. This is the wrong repository for the pitft. Please close this ticket here and open a new one in the related repository or a user group. Thank you

Andrysky, 2023-02-24:
@oldman4U 
HI, can you tell me where the required repository or user group is?
I can’t create a issue in [bigtreetech/OctoBTT](https://github.com/bigtreetech/OctoBTT) because the repository is in the archive.

p.s. i don't want to search for the company's contacts and write a private letter - as it may be useful to someone else.

oldman4U, 2023-02-25:
Sorry but I don't own this display.

Andrysky, 2023-05-02:
up

closed 2023-05-20 by stale[bot]