bigtreetech/BIGTREETECH-TouchScreenFirmware #2730: [BUG] not right extruder moving, if press button and turn knob in the same direction

svd05, 2023-02-22:
### Description

Not right extruder moving, if press button and turn knob in same direction.
Turning knob in CW it's always "+", CCW = "-". This is constant in many CNC machines and other devices.
If parameter "inverted_axis" set in "0", after then pushing button on the screen or turn knob a extruder move to the same direction (button, knob by CW). But if change "inverted_axis" on "1" and push button +X and turn knob CW, printer header will be move in different direction.
It looks like this:
[image]

### TFT Firmware Version & Main Board Firmware details

Motherboard SKR Pro V1.2
BTT TFT35 V.3.0
TFT Firmware Version 27.x
Marlin firmware version number 2.1.x Bug Fix.
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

digant73, 2023-02-23:
`inverted_axis` MUST be used according to the printer HW. E.g. you have to invert X axis if in your printer the X coordinate is increased when moving from right to left. In that case, the knob logic is also inverted (CCW will increase X (move to left)).
Same logic for Y and Z.

svd05, 2023-02-25:
Ok. Thank you for answer.
In my opinion and if we look at all CNC machines, this is very strange decision.
No matter where the plus of the x-axis is directed. Knob CW move to "+", CCW to "-". 
It is fact.

svd05, 2023-02-25:
And yet, how to disable the inversion of the rotation of the handle in the firmware.

closed 2023-05-08 by stale[bot]