bigtreetech/BIGTREETECH-TouchScreenFirmware #2731: [Q] Home offset and BabyStep

svd05, 2023-02-22:
**Your question or details you need. Please describe.**
Explain me please difference between these two values and what I should to save in EEPROM. I asking because i can't understand why when I move table to up "Home offset" has positive value (increment).
Logically, when table move up, then a clearance between table and nozzle decrement and BabyStep has negative value. Table can be moved beyond endstop. Thus, the offset can be negative with respect to zero.

Why is "Home Offset" positive if we make a negative shift and vice versa.

**The device variant related to my question.**
Motherboard SKR Pro V1.2
BTT TFT35 V.3.0
TFT Firmware Version 27.x
Marlin firmware version number 2.1.x Bug Fix.

[image]

digant73, 2023-02-23:
If you have a probe (e.g. BLTouch etc...), Marlin should communicate the presence of the probe to the TFT fw (with the reply to M503 sent by the TFT). If so (you have a probe), check (and fix) in your Marlin fw why Probe presence is not provided to the TFT fw.
If Probe presence is provided by Marlin, Probe Offset (and not Home Offset) value is displayed on Babystep menu.
If Probe presence is not provided by Marlin, Home Offset value is displayed on Babystep menu.
Home Offset logic is the opposite of Probe Offset (increased value instead of decreased value).

svd05, 2023-02-25:
So, if I right understand, I can't use M851, "BabySteps" and "Home offset" before installing BLTouch ?
Or if I can, what I should to save in memory from what i see on the display ?
BabySteps or Home Offset ? And wich sign to writing ?
Thx.

digant73, 2023-02-25:
babystep is a volatile value (it is not saved to EEPROM). Only the displayed value for Probe Offset or Home Offset are saved to EEPROM if you press on the `Save` button

svd05, 2023-02-26:
Finally, I understood this logic. What we offset relative to zero and what we moving.
1. What BabyStep this is a gap at the current time and sign indicates increase or decrease it. 
2. And Home Offset this is offset nozzle relative to zero.  Offset sign indicate direction offset relative axis direction.

After that, I can't understanding why changing icons (table instead of nozzle) when inverted Z-axis in BabyStep menu ? I mean only icons, not moves. Why are we doing this when we'are talking about nozzle displacement, not table displacement relative to the origin ?

I changed piece of code. Now logic and icons match each other.

[image]

digant73, 2023-02-26:
inverted Z-axis is typically used on printers were the moving part is the bed and not the nozzle

closed 2023-03-22 by bigtreetech