bigtreetech/BIGTREETECH-TouchScreenFirmware #2734: [Q] Artillery X2 and TFT35 problems with the filament sensor and other things

HaYaToSkAn, 2023-02-23:
**Your question or details you need. Please describe.**
Hello everyone
I have installed in my Artillery X2 with its original board (Artillery Ruby) a TFT 35 V3.0.1
I have updated Marlin to version 2.1.2
And the TFT with the file "BIGTREE_GD_TFT35_V3.0.27.x"
I have everything connected and working, but I have found some problems that I can't solve.

**Specify the device variant related to your question.**
Motherboard Artillery Ruby (Artillery X2)
BTT TFT35 V.3.0.1
TFT Firmware Version GD_TFT35_V3.0.27.x
Marlin firmware version number 2.1.2

I leave my Marlin and TFT files, as well as photos and videos.
I hope someone can help me
Thank you so much
PS: I have used the google translator, I hope you understand me well

**Additional context**
1. The filament sensor doesn't work, I've tried switching between the options in the TFT config file but nothing happens.
[image]
[image]
https://www.youtube.com/watch?v=ashsVZcDBPI

2. When I'm in TOUCH MODE printing and I want to switch to MARLIN MODE it doesn't allow me, it just doesn't give me the option.
On the other hand, when I print from MARLIN MODE and I want to go to TOUCH MODE it lets me.
In the TFT config file I have this option like this:
"serial_always_on:0"
https://www.youtube.com/watch?v=nbrtmp-bWVw

3. When I am printing from the MARLIN MODE option and I go to TOUCH MODE I only see the temperature status and something else, it does not give me the option to pause or cancel the print, to see how long it has been printing, the layer height and others... as if it were printing in TOUCH MODE
I don't know if this is normal or some configuration error.
[image]
https://www.youtube.com/watch?v=jL7amO7VS5w

Marlin and TFT Files 

https://drive.google.com/file/d/1UhWS3bAbs4pMcQXgoOs5CO9GaSROTp8n/view?usp=share_link

kisslorand, 2023-03-07:
1. As far as I know the filament runout sensor on the ASWX2 is an optical sensor which requires 5V to operate. As seen on the picture provided the TFT feeds only 3.3V to the sensor. Do you have the LED lit up? Does it change from green to red as you move in and out filament to/from it?

kisslorand, 2023-03-07:
As far as I know on the ASWX2 the [filament runout sensor](https://cdn11.bigcommerce.com/s-1vohxdh8gj/images/stencil/1280x1280/products/2444/7565/ART_Lswitch_02__34692.1657037545.jpg?c=2?imbypass=on) is an optical one which requires 5V to operate. In the picture you provided it can be seen that the TFT provides only 3.3V.
Does the LED on the sensor lights up? Does it change color from green to red as you move in and out filament to/from it?

HaYaToSkAn, 2023-03-07:
> 1. As far as I know the [filament runout sensor](https://cdn11.bigcommerce.com/s-1vohxdh8gj/images/stencil/1280x1280/products/2444/7565/ART_Lswitch_02__34692.1657037545.jpg?c=2?imbypass=on) on the [...]

This is the Original TFT that I removed
As can be seen the connector indicates 3.3V
The color it shows is always green, it never changes to red, I suppose it is because the sensor is not working correctly, due to some bad configuration in Marlin or in the TFT firmware.
Thanks for your help

[image]
[image]

kisslorand, 2023-03-07:
> As can be seen the connector indicates 3.3V

Mea culpa, I remembered it wrong.

freakydude, 2023-04-21:
The filament sensor works with 3.3V and 5V

Avatarsia, 2023-07-05:
i use the firmware on a stock tft and can confirm that it wont change anything in filament sensor logic when you change it in the config.txt
but it need to be changed

kisslorand, 2023-07-23:
@HaYaToSkAn

Check if the runout sensor works if you switch the two outside pins in one of the runout sensor cable connector or just try to insert the connector rotated 180°.

PR #2901 (2024-02-28): Bugfixes on top of last merged PR
**BUGFIXES:**
- Fixed #2895: Bug introduced by #2889 and affecting TFT variants with a LED on knob (e.g. TFT35 E3 V3.0 etc.) in case `knob_led_idle` setting on `config.ini` file was enabled. The bug caused a freeze exiting from PID/MPC menus or during the `Event LED` process, if enabled, at the beginning of a print. The bug was due to the missing `volatile` qualifier on `ms` attribute of OS_COUNTER data struct.
  **NOTE:** The bug severity can be considered major for the affected TFT variants
- Fixed #2898: Bug causing a freeze entering on `Preheat` menu. The  [... cut]
closed 2024-02-28 by bigtreetech