bigtreetech/BIGTREETECH-TouchScreenFirmware #2738: Beeping sound while installing on MKS Gen L V2.1 board [Q] (Your Question?)

dreed47, 2023-03-02:
**Your question or details you need. Please describe.**
I'm attempting to install the TFT35-E3 V3.01 on the MKS GEN L V2.1 controller board (w/ Marlin FW).  When I connect the EXP 1 and EXP 2 cables then the RS232 cable to the AUX connectors on the controller board I see the LCD first screen and a constant beeping tone and the touch screen doesn't appear to be working.  

**specify the device variant related to your question.**
TFT35-E3 V3.01
MKS GEN L V2.1 controller board
Marlin  with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER set on config

**Additional context**

radek8, 2023-03-03:
They sent a photo of the wiring.
Send configuration files.
Send a video with the problem.

oldman4U, 2023-03-06:
What is the status of the reported issue?

dreed47, 2023-03-06:
I will attempt installing this LCD one more time today and provide info.

oldman4U, 2023-03-06:
Follow the troubleshooting section of the Readme.

oldman4U, 2023-03-09:
Whats the status of the reported problem?

dreed47, 2023-03-11:
Unfortunately I have no further details. I was unable to get this unit to work and I needed to return the printer to its owner as I was only repairing it for them.

closed 2023-03-11 by dreed47