bigtreetech/BIGTREETECH-TouchScreenFirmware #2740: Mixing extruder

IoakeimSogiakas, 2023-03-06:
I have a weird issue. I have set up marlin correctly with a mixing extruder. In marlin mode, on the load/unload section, I have access to both extruders. However on tft mode I only have access to one. I have build my own firmware and I have defined 2 extruders and I have enabled the mixing extruder option however I still don't have access to the second one. 
Am I missing something? 

[image]

oldman4U, 2023-03-06:
When you press Nozzle?

IoakeimSogiakas, 2023-03-06:
Sorry I don't remember I will sent an update in about 2h (I am not home). From what I remember it didn't do anything but I might be mistaken

IoakeimSogiakas, 2023-03-06:
[Update]:it does nothing

kisslorand, 2023-03-06:
```
#### Hotend Count
#   Value range: [min: 0, max: 6]
hotend_count:1
```

What's the value set to in the "config.ini"?

IoakeimSogiakas, 2023-03-07:
It is set to 2

kisslorand, 2023-03-07:
Ah! I just realized you have one nozzle but two extruders. In this case I am afraid the TFT cannot handle more than one extruder per nozzle.

IoakeimSogiakas, 2023-03-07:
OK I will check it out thanks.

closed 2023-03-07 by IoakeimSogiakas

radek8, 2023-06-02:
In the configuration.h file, set:
#define EXTRUDER_COUNT 2
#define MIXING_EXTRUDER 1
compile your own FW for TFT

in the config.ini file set:
ext_count:2
upload configuration to TFT

IoakeimSogiakas, 2023-06-02:
The issue is fixed the extruder wasn't a mixing one so it didn't need the mixing functionality