bigtreetech/BIGTREETECH-TouchScreenFirmware #2742: Inverted Axes ini option problem[FR] (feature request title)

Grzesinski, 2023-03-06:
In my opinion behaviour of the inverted_axis option is inconsistent and should be tidied up.
I am particularly interested in getting only the reversal of the table move keys without changing the logic of the levelling operation.
It worked correctly in previous FW versions, but after entering the additional LY parameter, it didn't work any more.

Currently, the operating logic is as follows:
Y0 LY0 moving-normal levelling-normal [OK]
Y0 LY1 moving-normal levelling-inverted [OK]
Y1 LY0 moving-inverted levelling-inverted
Y1 LY1 moving-inverted levelling-inverted [OK]

The third option works exactly the same as the fourth.
I would expect action:
Y1 LY0 moving-inverted levelling-normal
so that the directions of the screen moving arrows coincide with the direction of the table movement, as it works for the other axes. This is the most intuitive operation of this key regardless of numerical changes to the Ypos value.

### Discussed in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/discussions/2735

digant73, 2023-05-12:
which menus are you referring to?

oldman4U, 2023-05-18:
@Grzesinski Please answer if you are still interested to get this done. Otherwise please close this ticket. Thank you

Grzesinski, 2023-05-18:
The listed entries in the ini file affect the behaviour:
Y [0 or 1] the direction of manual Y-axis movement, Menu  Movement -> Move -> Y(+-)
YL [0 or 1] the direction of movement of the Y axis during bed leveling, Menu Movement - Leveling

Option Y1 LY0 means that only manual moving direction is inverted, Y direction at bed leveling works normally.
Now affect also bed levelin direction and works same as Y1 LY1 option.
I can't change manual moving direction only.

digant73, 2023-05-18:
@Grzesinski the bug has been introduced by #2655 that reported as a bug the proper logic (the one you reported) fixed by #2233. Something that is constantly happening since a lot of time now.

Grzesinski, 2023-05-20:
I downloaded the latest (from today) version of FW.
At first glance, the descriptions in the ini file are the same as before.
Are you sure changes have been made?

"We separate a single variable "LY" (Leveling Y axis) to deal with the Y axis leveling movement direction.
Format: [X<option> Y<option> Z<option> LY<option>]
Options: [disable: 0, enable: 1]
inverted_axis:X0 Y0 Z0 LY0"

But if only one value is currently valid for the Y-axis, it basically doesn't change much.
As I stated earlier:
The current Y0 (formerly Y0 LY0) was working fine before.
The current Y1 (previously Y1 LY1) works fine as well.

I need the Y1LY0 configuration and I will not get it on any of them. In both of these versions, Y1 overrides LY to 1.

Should I check the operation of the new version?

digant73, 2023-05-21:
The fix is not yet merged in the official TFT fw. It is in PR #2792. That PR is fixing the bug (so LY<x> usage will be restored on Leveling menus while X<x> Y<y> Z<z> params will be used on Move menu).
No, there is no need you download/test the PR. I just asked you to check on your current TFT if also "Y0 LY1 moving-normal levelling-inverted" was KO (it should be KO (no inverted leveling). You reported that it was OK, but this is not possible on current fw (due to the bug).

Grzesinski, 2023-05-21:
You're absolutely right.
My mistake. 
Y0LY1 - Y not inverted, LY not inverted.
also
Y1LY0 - Y inverted, LY inverted.

So Y value directly overwrite LY.

closed 2023-05-24 by bigtreetech