bigtreetech/BIGTREETECH-TouchScreenFirmware #2743: [BUG] TFT70 only levels a quater of the bed

Kreation88, 2023-03-07:
### Description

I have an Ender 5 plus with an SKR 1.4 turbo and a TFT70 that I've recently upgraded to from a TFT35. When I setup the TFT70 I upgraded to the latest firmware. When I try and manual level the bed the printer thinks the bed is 25% of its actual size and moves to the front left section of the bed for tuning the z offset. I can still use the auto level to probe the bed and that does the correct grid on the full bed and I can print on the whole bed. I've tried the previous firmware with the same results. But if I hook my TFT35 back up it goes to the four corners of the bed as it should and sets the Z offset in the middle of the bed. I've also tried re flashing my firmware to my SKR 1.4 and a hard reset on the TFT70 with no luck either.

### Steps to reproduce

1. [First Step] - Try to manually level the four corners.
2. [Second Step]
3. [and so on...]

**Expected behavior**
 Go to the corners of the bed at first layer printing height to level the bed manually/

**Actual behavior**
 The printer thinks 

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details BIGTREE_TFT70_V3.0.27.x and main board firmware-btt-skr-v14-turbo-e5p-2.0.6-dual-Z-2209-20200804

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Kreation88, 2023-03-15:
Figured this out the config file has the max bed size set to and 235mm2 for some reason, increased it to my actual be size and all is good. WHY IS THE SCREEN SETTING THE MAX BED SIZE????? Should that be the main board firmware???? Its like your can stereo making it so you can only go a restricted space. MAKES NO F#$KING SENSE...

kisslorand, 2023-03-15:
It makes sense...
Some users are so ignorant, they do not even bother to read the "readme" section but they spend time renting.  The TFT will auto-detect the machine size (min and max) as it is defined in Marlin firmware (requires enabling "M115_GEOMETRY_REPORT" in Configuration_adv.h in Marlin firmware).

Is "M115_GEOMETRY_REPORT" enabled in "Configuration_adv.h" in your printer's Marlin firmware?

Kreation88, 2023-03-15:
You’re ignorant as well… I have a TFT35 that worked with no issues with the same firmware on the screen and on my main board and I read a lot on here trying to figure this out with no replies for a week I did some more digging and found the fix. No matter how many times I re flashed firmware or reset with TFT70 it still limited by bed size but the TFT35 had no issues. Definitely something is off.

oldman4U, 2023-03-16:
@Kreation88 Do you understand that this is not a user group but the BTT TFT firmware repository? Maybe using a user group you would have got the answer faster. Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2023-03-16:
How did I attack you?

kisslorand, 2023-03-17:
> I sent the M115 on the TFT terminal and didn't see what you listed.

Is "M115_GEOMETRY_REPORT" enabled in "Configuration_adv.h" in your printer's Marlin firmware?

Duh!

kisslorand, 2023-03-17:
Do you understand that Marlin has to have "M115_GEOMETRY_REPORT" enabled so it would report the bed size to the TFT?
Do you expect the TFT to guess the size of the bed???

I guess you do.... Pffff...

Kreation88, 2023-03-17:
> Do you understand that Marlin has to have "M115_GEOMETRY_REPORT" enabled so it would report the bed size to the TFT? Do you expect the TFT to guess the size of the bed???
> 
> I guess you do.... Pffff...

Like I said before for the third time at least the same marlin firmware worked with the TFT35 and knew the size of my bed, and the TFT70 knew the size of the bed to create a mesh with a BL touch but not went back to the default setting in the config file for bed size....

kisslorand, 2023-03-17:
OK, good luck!

Kreation88, 2023-03-17:
> OK, good luck!
> 
> > So the TFT70 knew my bed size for creating a mesh but not for manually leveling the four corners?
> 
> My last answer on this topic: No, the TFT just gave a G29 command, it was Marlin who knew the bed size for creating a mesh. For manual leveling the TFT has to know the bed size, because it is the TFT which calculates the corner coordinates based on the known bed size.

But then why did the TFT35 work with the same marlin firmware?

digant73, 2023-05-19:
@Kreation88 if not too late, on the TFT35 (you reported a working) did you check the value set for parameter "machine_size_max" in config.ini? if it matches the value you needed to set for TFT70 then if you set it to a wrong value (e.g. half the correct bed size) you should have the same issue as in TFT70. Obviously you have to make the tests connecting both TFTs to the same main board (Marlin)

closed 2023-09-17 by stale[bot]