bigtreetech/BIGTREETECH-TouchScreenFirmware #2745: filament is over

Vladiimirs, 2023-03-08:
I use the latest firmware 9.3 tft 28 artillery x1 marlin 2.1.2

After flashing the display and resetting the settings to factory settings. I turn on the filament sensor and when I press print, the printer says that the filament is over. remove the filament from the sensor, printing continues.
changed in firmware
#### Inverted Filament Runout Sensor Logic
# The sensor uses an inverted logic.
# Disable it in case filament runout is triggered by a LOW signal.
# Enable it in case filament runout is triggered by a HIGH signal.
# Options: [disable: 0, enable: 1]
fil_runout_inverted:1
Nothing changed

changed parameter
#### Filament Runout Sensor
# Select the type of filament runout sensor and its default enabled/disabled state.
# Options: [Normal Disabled: 0, Normal Enabled: 1, Smart Disabled: 2, Smart Enabled: 3]
fil_runout:1
nothing happened either

tried
fil_runout_inverted:0
fil_runout:1
also nothing

changed
#### NC (Normal Close) Filament Runout Sensor
# The sensor is of type NC.
# Options: [Normal Open: 0, Normal Close: 1]
fil_runout_nc:0

and also nothing

What to do, you really need a filament sensor.

Vladiimirs, 2023-03-08:
everything was resolved, everything was spoiled by reset.txti in the included m600

oldman4U, 2023-03-14:
@Msq001 please close this ticket

closed 2023-03-22 by bigtreetech