bigtreetech/BIGTREETECH-TouchScreenFirmware #2746: [Q] Print job after stopped print job always fails

rondlh, 2023-03-12:
After I cancel a print by pressing "Stop" on the TFT, the print stops successfully, but the next print always fails spectacularly. 
After the stopped print, when I start a new print, the homing, heating and bed leveling works correctly, but then the print head moves to the top right corner (XY kinetics) and bumps into the sides and start printing (outside of the bed). Normally it should go to the center of the bed to print. 

**Could somebody confirm if they see the same problem? Please let the print you want to cancel proceed to the actual printing of the part before stopping.**

**Configuration:**
BTT TFT35 V3.0 on latest FW, Marlin FW (Bugfix March 11th 2023, XY kinematics).

kisslorand, 2023-03-12:
I started a print from TFT USB and stopped the print whilst printing the skirt by pressing the "Stop" button. The print stopped, I cleaned the bed, went to main menu and than started again the same print. Nothing unusual happened, the print started as it should. In fact it is still ongoing just fine.

rondlh, 2023-03-20:
I will investigate further

closed 2023-03-20 by rondlh