bigtreetech/BIGTREETECH-TouchScreenFirmware #2748: illegal user app

epicgammer12, 2023-03-15:
When I start up the Printer the screen displays illegal user app and nothing else

I installed the latest firmware, config.ini, lanuage_eu.ini, and the tft 35 folder.

I have tried everything including looking through other similar problems (like illegal flash app)

Note that when I install new firmware the touch screen works and everything boots up normally, however when I turn off the printer then turn it back on again it goes directly back to illegal user app. 

(the exact behavior it the screen is black besides 3 words in the center which are white that read "illegal user app!"

kisslorand, 2023-03-15:
You forgot to mention very important details: the EXACT version of the TFT you have and the name of the file you flashed it to.
Also worth mentioning that you should try a reset by creating an empty "reset.txt" file on your SD card and leaving that card in the TFT SD slot upon booting.

epicgammer12, 2023-03-15:
I am using a bigtreetech tft35-e3 v3.0.1 with a GD chip

and the file I flashed it with is called "BIGTREE_GD_TFT35_V3.0_E3.27.x.bin"

epicgammer12, 2023-03-15:
Also I tried the reset.txt thing and it did not work

kisslorand, 2023-03-15:
I also checked and the message "illegal user app!" is not present in the firmware so it must be in the bootloader. In that case it's up to @bigtreetech to give a proper response or to come up with a fix since the bootloader is not publicly available.

epicgammer12, 2023-03-15:
oh ok then

epicgammer12, 2023-03-15:
thanks for the help though

epicgammer12, 2023-03-15:
Update, I changed nothing but it magicly started working so idk what happended but it fix itself so I am going to close this.

closed 2023-03-15 by epicgammer12

yourbuddydinec, 2023-03-25:
I'm having the same problem today. Same error, same screen, same firmware

yourbuddydinec, 2023-03-25:
I may have figured it out if anyone else gets this problem. I think it was a problem with the download from github. I noticed on the config.ini it was a larger file size and had a bunch of extra trash text in there. I redownloaded it all "raw" and it worked.