bigtreetech/BIGTREETECH-TouchScreenFirmware #2749: [Q] Marlin Mode Fails to Work SKR Mini E3 V3 board and TFT24 v1.1

helakejr, 2023-03-16:
I upgraded the SKR Mini E3 V3 board and TFT24 v1.1 display at the same time and the Touch Mode screen works fine however when selecting Marlin Mode, the screen remains black except for the "Marlin Mode" title. After much testing, I discovered both the Marlin and touch screen modes work if I connect the serial cable to the SWD port on the TFT24. 

Everything works fine on the Marlin Mode in Touch Mode I get "No Printer Attached" message on the touch screen. 

Marlin Configuration configured per :
#define SERIAL_PORT 2		// Host port
#define BAUDRATE 115200

#define SERIAL_PORT_2 -1
//#define BAUDRATE_2 250000   

In summary, when I connect:

BTT SKR Mini E3 V3	TFT24 v1.1
10-pin EXP1		to	TFT24 EXP1/EXP2 split cable
5-pin serial	to	TFT port	Touch mode works, Marlin Mode blank Screen
5-pin serial	to	SWD port	Touch mode does not connect, Marlin Mode works

Test Setup:
10-pin EXP1	to	TFT24 EXP1/EXP2 split cable
5-pin serial	to	TFT port	Touch mode works, Marlin Mode blank Screen
SKR Mini E3 V3 board jumpered to provide +5V - Touch mode works, Marlin Mode blank Screen
SKR Mini E3 V3 board external +5V  - Touch mode works, Marlin Mode blank Screen

I have replaced both: BTT SKR Mini E3 V3, and TFT24 V1.1 with same results.

radek8, 2023-03-18:
Never connect TFT and SKR via the SWD connector. You risk destroying the pins that use 3.3V logic. The TFT and SWD connectors have the power pins reversed.

The problem will be in the connection of your split cable
Can you send the wiring diagram of your cable?

helakejr, 2023-03-18:
This cable was provided by BIQU with the TFT24 v1.1.

[image]
[image]

radek8, 2023-03-18:
The picture is correct. Check if the cable is connected as well as the picture..
Check the light blue lines.

helakejr, 2023-03-18:
I checked the cable. It is built correctly. I replaced the SKR mini e3
board and the TFT24. The TFT24 had the same problem. And the cable was
built the same way.

radek8, 2023-03-18:
Send photos of the cable plugged in.

helakejr, 2023-03-18:
With only the 10-pin split cable connected - the TFT24 reset - resets both. However, still Marlin Mode header with blank screen.

radek8, 2023-03-18:
I have a question.
1. If you connect only the serial cable (5-pin), does the touch mode work and can the printer be controlled?
2. If you connect only a 10-pin cable, does the Marlin mode work for you?
3. If you connect both cables, the Marlin mode does not appear, but the touch mode controls the printer?

helakejr, 2023-03-18:
[image]

1. Yes, with only the 5-pin connected the touch mode works and the printer can be controlled
2. No, Marlin Mode header blank screen
3. Yes, the touch mode works and controls the printer, Marlin Mode header only

[image]
[image]

helakejr, 2023-03-18:
Question:
What serial port does the SWD port use?

radek8, 2023-03-18:
The SWD connector is not a serial port!
Compiling your own Marlin?
Send configuration files.

helakejr, 2023-03-18:
I used the prebuilt with the same results.

Here is my build.
[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11009146/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11009147/Configuration_adv.h.txt)

radek8, 2023-03-18:
The display driver is fine

radek8, 2023-03-18:
Did you connect the 5-pin cable to the SWD connector on the display or on the SKR board?

helakejr, 2023-03-18:
5-pin connected to SKR TFT connector.

helakejr, 2023-03-18:
[image]

radek8, 2023-03-18:
The EXP2 connector is connected incorrectly.
It is shifted by one pin!

[image]

radek8, 2023-03-18:
Connect the EXP2 connector correctly and hope you didn't destroy anything connected to the SWD connector

helakejr, 2023-03-18:
Still doesn't work.

helakejr, 2023-03-18:
[image]
[image]
Still doesn't work.

helakejr, 2023-03-18:
> Did you connect the 5-pin cable to the SWD connector on the display or on the SKR board?
> The other end of the cable remained in the TFT connector?

To get it to work, left TFT 5-pin on correct connector and plugged into the SKR SWD connector.

helakejr, 2023-03-18:
I don't know if it matters or not but my TFT24 has a GD cpu.

radek8, 2023-03-18:
> I don't know if it matters or not but my TFT24 has a GD cpu.

CPU type has no effect

radek8, 2023-03-18:
> > Did you connect the 5-pin cable to the SWD connector on the display or on the SKR board?
> > The other end of the cable remained in the TFT connector?
> 
> To get it to work, left TFT 5-pin on correct connector and plugged into the SKR SWD connector.

How was the connector oriented on the SWD connector?

helakejr, 2023-03-18:
Reversed from the TFT24 serial port.

radek8, 2023-03-18:
So you linked together
reset with reset a
5V with 3.3V?

helakejr, 2023-03-18:
Don't know for sure as I haven't repeated it since your admonishment. But I
believe so.

radek8, 2023-03-18:
All reset pins are connected together.
By connecting the TFT and SWD connectors, the Reset +5V +3.3V connections were made. The reset and 3.3V buses were powered by 5V.

radek8, 2023-03-18:
a similar problem

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/964

radek8, 2023-03-18:
If touch mode works for you, you don't really need Marlin mode.
I haven't used the Marlin mode on the printer for about 2 years :-)

helakejr, 2023-03-19:
I believe that this is the solution. Touch mode and its graphical user
interface.

Thank you

helakejr, 2023-03-19:
Yes

helakejr, 2023-03-19:
I believe that since Marlin Mode doesn't work and Touch mode is so much
better. I will just use the 5-pin serial connection and save running both
cables.

Thank you for the help.

closed 2023-03-19 by helakejr