bigtreetech/BIGTREETECH-TouchScreenFirmware #2750: [Q] BTT TFT 3.5 V3.0.1 compatibility?

rondlh, 2023-03-17:
I just noticed version 3.0.1 of the BTT TFT 3.5 (https://biqu.equipment/products/btt-b1-tft35-v3-0-display-touch-screen-two-working-modes-for-biqu-b1), but unfortunately no update on the Github page (https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0) so I'm not sure if it is supported by the software hosted here. Does anybody know what's new/changed? Is it supported?

I hope they fixed the conflict in the used PWM channels for the display backlight LED and the speaker, which is very inefficient. Now sounds are created by generating a train of interrupt at double the sound frequency, but using PWM would be much better (more accurate and flexible) and less resource intensive.

radek8, 2023-03-18:
Yes, the B1 version is compatible with this repository.
Attention! It comes in two variants, ST and GD.
You must select the correct FW variant when updating.

The difference is in the layout of the connectors and the CPU version.

[image]

radek8, 2023-03-18:
If the display processor is labeled GD32...
use FW with index GD
(The version of the board is usually marked as V3.0.1)

If an STM32 processor is used...
use FW without index
(The version of the board is usually marked as V3.0)

rondlh, 2023-03-19:
> If the display processor is labeled GD32... use FW with index GD (The version of the board is usually marked as V3.0.1)
> 
> If an STM32 processor is used... use FW without index (The version of the board is usually marked as V3.0)

Thanks a lot, that's good to know.

rondlh, 2023-03-20:
Thanks for the clarification

closed 2023-03-20 by rondlh