bigtreetech/BIGTREETECH-TouchScreenFirmware #2753: Resolving the 'No Printer Attached" error with BTT SKR mini E3 V3.0.1

keshka, 2023-03-19:
I purchase the set, a new SKR Mini E3 V3.0 Control Board +TFT35 E3 V3.0.1 Touch Screen Display 32-Bit. I followed this video for connection. 
https://www.youtube.com/watch?v=dtwWnz2-5VQ&ab_channel=AuroraTech
I flashed the [THEME_The Round Miracle Menu Material theme](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_The%20Round%20Miracle%20Menu%20Material%20theme)
[BIGTREE_GD_TFT35_V3.0_E3.27.x.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0_E3.27.x.bin)
[config.ini](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini)
and 
[language_en.ini](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/Language%20Packs/language_en.ini)
to a SD card and flashed the display board. The unmodified configuration file indicates a serial speed of 250000 but on first checking using the touch screen menu "settings" the P1 serial port was set to 115200. However, trying  other speed settings has no effect.

attempting to trace down this problem I verified the TPT pinouts vs the pinouts of the SKR Mini E3 V3.0 Control Board
[image]
[image]

The supplied cable was incorrect. I rewired it as such:
TFT         SKR
5v     to    5v
GND to  GND
TX     to   RX
RX     to   TX
RST   to   RST

I confirmed the cable integrity by conducting a continuity test from the solder points on the TFT board and solder points of the SKR board. All check out.

The display works fine in the 'Marlin' mode

How do I confirm the serial speed of the SKR? It has factory firmware on it at this point. 

If it needs to be flashed, please direct me to the appropriate GITHUB repository. This one referenced by the BTT web site: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3
 only has [V3.0.1 STM32F401 version] ....mine has the STM32Gxxx processor.

radek8, 2023-03-19:
Check the wiring according to this picture.
Your image is from a display perspective. So it is mirrored and can be confusing.

[image]

radek8, 2023-03-19:
The order of pins on the serial cable is the same for both SKR and TFT.
You just have to orient it correctly according to the reset pin and 5V
Do not change the pin order.
Send a photo of the connection.

You can find out the speed of the serial port in Marlin from the menu
About the printer
info about the board.

radek8, 2023-03-19:
You are inattentive.
The raw configuration file sets the speed to 115200

[image]

radek8, 2023-03-19:
Try this but it's a year old version.
But you can use it as a template to build your own new FW.

https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1

keshka, 2023-03-19:
radek8; 
Thank you for the response. I will use your pinout photo and double check wiring. Your right, tx to tx is not standard, I will flip them from tx-rx back to tx-tx, ect. I see why I missed the speed in the configuration file. It looks like 0, 2400 rather than OFF 0. That verifies the speed should be 115200 then. I have work to do! 

Off topic question, do you know how to use an inductive Z probe with the SKR?

radek8, 2023-03-19:
I have no experience with an induction probe. I use 3DTouch

oldman4U, 2023-03-22:
Keshka. What is the status of the reported issue? Please help the community and close the ticket once you do not need it anymore. Thank you

keshka, 2023-03-22:
it remains un-resolved. I will attempt a manual build and flash of the op system to see if that resolves the problem. I used an Arduino to verify which pins are TX vs RX on the screen and the SKR as well as the correct baud rate. My connection was correct and set speed was correct. It appears the screen is just not paying attention. The SKR sends out a stream of numbers but the screen seems to ingore the data. I originally re-flashed the screen but the SKR seems to be running software installed by www.th3dstudio.com

radek8, 2023-03-27:
You have to enable serial port 2 on the SKR
Do you have this set in the SKR configuration?

radek8, 2023-03-27:
[image]

QUADMAN7, 2023-03-27:
I am having a similar issue on my printer with a bigtreetech Manta m5p running klipper. My TFT 35 V3. 0.1 says there is no printer attached. I am using the GPIO pins on the board and I have connected the board to those pins using the black cable that came with it. Does anyone know how to resolve this no printer attached issue?

keshka, 2023-03-28:
Since the display will not talk to the SKR, is reflashing it the only way to turn serial 2 port on? 

In addition, I belive the port is turned on already. As mentioned above, when attaching an Arduino to the "rs232" pins on the SKR, I can read a stream of numbers at 115200 so it appears the SKR is sending data, the screen is not listening.

radek8, 2023-03-29:
> Since the display will not talk to the SKR, is reflashing it the only way to turn serial 2 port on?
> 

If serial port 2 in Marlin is not active then Yes.

> In addition, I belive the port is turned on already. As mentioned above, when attaching an Arduino to the "rs232" pins on the SKR, I can read a stream of numbers at 115200 so it appears the SKR is sending data, the screen is not listening.

Check the serial cable if it is not faulty on the RX TX pins.
It has happened several times that the display had a faulty pin on the RS232 connector.
Users solved it by modifying the FW for the display and switching the RX TX pins from the RS2320 connector to the Uart3 connector.
If the RS232 was faulty, this would be a solution for you as well.

keshka, 2023-03-29:
That is a plan, I will try that and post results

oldman4U, 2023-04-07:
Keshka. Any news?

keshka, 2023-04-07:
When this project failed, I began work on a Voron 2.4. It's nearly finished. When it is, I can get back to this. Thank you for following up.

Hep43, 2023-04-11:
is the issue resolved? im having the same problem with ender 3p skr mini e3v3 and tft35

closed 2023-07-19 by stale[bot]