bigtreetech/BIGTREETECH-TouchScreenFirmware #2754: BTT TFT35 E3 V3.0 Mixing hotend issue???

Services8810, 2023-03-20:
Hi all,

I have a SKR 1.4 T with two extruders and Mixing hotend. I have compiled my mainboard firmware with these settings.

In the Configuration.h file
set the parameter
#define MIXING_EXTRUDER 1
#define EXTRUDER_NUM 2

At the TFT firmware, I have changed the config.ini file in order to include my 2 extruders, excatly as  "Mixing Extruder" #1055 issue describes. 

set a parameter in the config.ini file
ext_count: 2

I had the same problem, and I could not exchnage extruders in order to feed the filament, so I followed the same article and commected the following lines

[image]

The only change is that my parseACK.c files is newer and the lines that I commented were not at 584-588 but at 1164 - 1170.
After that changes I could exchange extruders and prime the filament but as  Option for Mixing_extruder #1123 arcticle describes at the printer name I have an unecessary line with the following

EXTRUDER_COUNT: 1 UUID: cede2a2f-41a2-4748-9b12-c55c62f367ff

I tried to use the file radek8 created by it seams very outdated as it is way smaller than the new file I have.

Can anybody fix this? 

Thanks, 
John

radek8, 2023-03-20:
Yes

Services8810, 2023-03-20:
Thank you very much. Should I change also #define Extruder_Count 2 ??? Default was 1

radek8, 2023-03-20:
Yes you can, it becomes the default value when resetting to factory settings. But the value that will be in the Config.ini file will overwrite this value.
It is important to define the number of ectruders in the Config.ini file and upload this configuration to the TFT.

Services8810, 2023-03-20:
Thank you once again! I will try these tomorrow.

Services8810, 2023-03-21:
It is working great. Thanks radek!

closed 2023-03-21 by Services8810

taraxias, 2023-08-24:
please someone tell me what i am doing wrong.. i am going to brake it...
i have made a custom printer with 3 steppers and a mixing hotend.. everything works fine,it prints fine exept that in touch mode i cant change between steppers to change filament. it only "reads" 1 stepper. i follow almost everything i have found in here and nothing,5 days now i cant fix it.. pleaseeeeee help. my motherboard is octopus v1.1 and tft35 e3 v3

[image]

[image]

[image]

[image]

[image]