bigtreetech/BIGTREETECH-TouchScreenFirmware #2755: Filament sensor runout

Blisk, 2023-03-24:
I have a motherboard SKR mini 3.0 and TFT35 display.
Now I have connected my filament sensor on SKR mini and it detects the sensor but still gets an error filament runout.
I have googled about it and found out I need to connect the filament sensor to TFT35 and not SKR mini.
Is there any schematic of how it should be connected? I measured all pins 2 have 3.3V and one is ground.
As I read I also need to upload firmware again and change settings in config.ini for the filament sensor.
Which pin is for data?

kisslorand, 2023-03-24:
Check the documentation of your TFT, the data pin is the one that is not labeled "GND" or "3.3V".

Blisk, 2023-03-24:
Thank you I missed that.

closed 2023-03-24 by Blisk