bigtreetech/BIGTREETECH-TouchScreenFirmware #2757: BTT SKR 3 EZ / TFT35 V3 no Printer Attached problem

Tomzrk, 2023-03-27:
I Recently changed the board on my ender 5 plus and got SKR 3 EZ with TFT35 v3 , I got the No Printer Attached and when i switch to marlin mode i got a black screen with nothing on it , I changed the Serial port on Marlin few times but nothing helped , i connected that screen to a different 3d printer and it was working fine so i swapped the connector between the 2 machines but its still the same , swapped the RX and TX nothing changed so i was wondering if i ![image](https://user-[image]images.githubusercontent.com/[image]113375679/227921461-492e57fb-d312-4d48-88fc-336b59039642.jpeg)missed something or its a faulty board[image]

Tomzrk, 2023-03-27:
[Marlin FW.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11077860/Marlin.FW
[BIGTREE_GD_TFT35_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11077876/BIGTREE_GD_TFT35_V3.0.27.x.zip)
.zip)

Tomzrk, 2023-03-27:
[image]

Tomzrk, 2023-03-27:
[image]

Tomzrk, 2023-03-27:
[image]

Tomzrk, 2023-03-27:
[image]

Tomzrk, 2023-03-27:
@radek8 hey Radek ,any more informations needed ?

Tomzrk, 2023-03-27:
@radek8 I had them both connected while testing but I took them off after i was done

closed 2023-03-28 by Tomzrk

radek8, 2023-03-28:
If the display works on another printer, then the problem will be on the SKR.
When uploading FW to SKR, will the file extension be renamed to CUR?
If you connect the SKR to a PC via USB, does the PC communicate with the printer?

Tomzrk, 2023-03-28:
I found the issue in the marlin firmware , i had the wrong envs selected 

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32H743Vx_btt
include_dir  = Marlin 

Instead it should be 

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32H723Vx_btt
include_dir  = Marlin

But now I'm having an issue with Z stepper not working when connected to Zam and Zbm

Tomzrk, 2023-03-28:
That's another issue I'm facing too , when I connect the board to Pc through USB nothing happens